arduino/Arduino #9596: Please add HARDWARIO SoilSensor library

hubmartin, 2020-01-06:
https://github.com/bigclownlabs/SoilSensor

cmaglie, 2020-01-07:
Done.

closed 2020-01-07 by cmaglie

hubmartin, 2020-01-10:
Thanks, just a question:
is it necessary to name the repository exactly asi is the name of the library? Could the repo be renamed for example `bcf-arduino-soil-sensor`?
Thanks

per1234, 2020-01-10:
@hubmartin as far as Library Manager goes, you can name the repository any way you like. The only name restrictions are on the `name` value in library.properties.

However, there is a benefit to having the repository name and library.properties `name` value both match the filename of the primary header file. The reason is that, when multiple libraries are installed that contain a file matching an `#include` directive, one of the factors in which library gets priority is the match between the library folder name and the filename. 

Library Manager installs the library to a folder named according to the library.properties `name` value (with any spaces changed to underlines), so the repository name doesn't matter for that installation method. However, if the user does an "Add .ZIP Library" or a manual installation, the library folder is likely to be named according to the repository name.

hubmartin, 2020-01-10:
@per1234 Thanks for quick and detailed answer :+1: I've expected it could be possible but thanks for confirmation.