arduino/Arduino #9600: Win32 Arduino IDE 1.8.10

sedlacekdavid, 2020-01-06:
Hi, I'm getting an error message when trying to compile my project. Moved recently to the latest version from the previous one.

`panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x7855fb]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.ResolveLibrary(0x11cda5a0, 0x11e500a1, 0x8, 0x11e500a1)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/resolve_library.go:64 +0x14b
github.com/arduino/arduino-cli/legacy/builder.findIncludesUntilDone(0x11cda5a0, 0x11d3a900, 0x88ed00, 0x11c8ae40, 0x11e33ba8, 0x11ffe800, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:358 +0x3b6
github.com/arduino/arduino-cli/legacy/builder.(*ContainerFindIncludes).Run(0xdebc68, 0x11cda5a0, 0xdebc68, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:152 +0x3e5
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x11cda5a0, 0x11c2fe1c, 0x20, 0x20, 0x409a01, 0x7fff, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0x11c71f40, 0x11cda5a0, 0xdebbb4, 0x11c42c88)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:117 +0xb9c
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:226
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:398 +0x6d2
C:\Program Files (x86)\Arduino\arduino-builder vrátil 2
Nastala chyba při kompilaci u desky Arduino Due (Programming Port).
`

sedlacekdavid, 2020-01-06:
This is an output from VS Code:

`[Starting] Verify sketch - sketch_main.ino
[Warning] Output path is not specified. Unable to reuse previously compiled files. Verify could be slow. See README.
Set log4j store directory C:\Users\David\AppData\Local\Arduino15
Nahrávam konfiguraci...
Inicializace balíčků...
DEBUG StatusLogger Using ShutdownCallbackRegistry class org.apache.logging.log4j.core.util.DefaultShutdownCallbackRegistry
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger AsyncLogger.ThreadNameStrategy=UNCACHED (user specified null, default is UNCACHED)
TRACE StatusLogger Using default SystemClock for timestamps.
DEBUG StatusLogger org.apache.logging.log4j.core.util.SystemClock does not support precise timestamps.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Took 0,208128 seconds to load 209 plugins from sun.misc.Launcher$AppClassLoader@af3868
DEBUG StatusLogger PluginManager 'Converter' found 44 plugins
DEBUG StatusLogger Starting OutputStreamManager SYSTEM_OUT.false.false-1
DEBUG StatusLogger Starting LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@dd7b39]...
DEBUG StatusLogger Reconfiguration started for context[name=af3868] at URI null (org.apache.logging.log4j.core.LoggerContext@dd7b39) with optional ClassLoader: null
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger PluginManager 'ConfigurationFactory' found 4 plugins
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Missing dependencies for Yaml support, ConfigurationFactory org.apache.logging.log4j.core.config.yaml.YamlConfigurationFactory is inactive
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Using configurationFactory org.apache.logging.log4j.core.config.ConfigurationFactory$Factory@1c19a70
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.properties] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-test.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.properties] using ClassLoader.getSystemResource().
[... cut ...]
DEBUG StatusLogger Shut down OutputStreamManager SYSTEM_ERR.false.false, all resources released: true
DEBUG StatusLogger Appender Console stopped with status true
TRACE StatusLogger XmlConfiguration stopped 2 remaining Appenders.
TRACE StatusLogger XmlConfiguration cleaning Appenders from 2 LoggerConfigs.
DEBUG StatusLogger Stopped XmlConfiguration[location=jar:file:/C:/Program%20Files%20(x86)/Arduino/lib/pde.jar!/log4j2.xml] OK
DEBUG StatusLogger Stopped LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@dd7b39] with status true
[Error] Exit with code=1

`

cmaglie, 2020-01-07:
This issue has been fixed on the latest `arduino-builder` that is still not released (it will be with the Arduino IDE 1.8.11).

In the meantime you can try with the hourly builds here: https://www.arduino.cc/en/Main/Software

PaulStoffregen, 2020-01-07:
@cmaglie - Any idea on the time frame for the 1.8.11 release?  I've been hearing a number of user complaints, so eager to see everyone move to a better build experience...

sedlacekdavid, 2020-01-08:
OK guys, thank you for the information. No problem, will wait then.

If this is already been addressed, I guess it can be closed.

closed 2020-01-22 by per1234