arduino/Arduino #9602: [Library Manager]Please add library

HANBACK-HALEY, 2020-01-08:
Hi, please add this library to the library manager.

https://github.com/HANBACK-HALEY/Hanbackpop

Thank you.

per1234, 2020-01-08:
In order to be added to the Library Manager index your library must contain a library.properties file. The format of this file is documented here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

After adding the library.properties file, you will need to create a new tag/release or redo the existing one. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

HANBACK-HALEY, 2020-01-08:
Hi, i added library.properties file. please check and add library. thank you.

HANBACK-HALEY, 2020-01-09:
please cancel this library. sorry.

closed 2020-01-09 by per1234