arduino/Arduino #9608: BeagleBone problems with create.arduino.cc

dacobi, 2020-01-10:
Hi 

I'm a newcomer to Arduino having used BeagleBone in the past.

Just installed the Arduino Create Agent and added a BeagleBone in the web interface/editor.

I can compile code and it runs on the BeagleBone but no hardware activity seems to happen.

I've tried a simple blink example and trying to write to the serial monitor but nothing happens.

Blink Example:

int LED_BUILTIN = 26;

void setup() {
    pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
    digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
    delay(1000);                       // wait for a second
    digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
    delay(1000);                       // wait for a second
}
 
I can change the value of GPIO26 in sysfs manually but nothing happens when the sketch runs.

Is this a bug or am I missing something entirely?

Is there any documentation specifically for using the BeagleBone with the Arduino web editor?

per1234, 2020-01-10:
According to https://github.com/arduino/arduino-create/issues/1#issuecomment-351344954, the [Create > Editor section of the Arduino Forum](http://forum.arduino.cc/index.php?board=101.0) is the appropriate place to submit bug reports or feature requests related to the Arduino Web Editor. I see you already have posted to that board (https://forum.arduino.cc/index.php?topic=658139) so I'll go ahead and close this.

closed 2020-01-10 by per1234

dacobi, 2020-01-10:
I'm at a loss here. You could't just tell me if the BeagleBone is supposed
to work with normal Arduino sketches er not?

matthijskooijman, 2020-01-15:
@dacobi, this is really the wrong place for that question. To prevent clutter (there are already enough issues) and extra spam for people (there are hundreds of people subscribed to this repository), we are quite strict in closing off-topic discussions and questions. In this particular case, I do not even know the answer to your question and I suspect @per1234 does not either.

dacobi, 2020-01-15:
The thing is, everywhere I ask this question I'm told its the wrong place.

Yet it is listed as a supported platform, and no one within or outside the
arduino project knows anything about it.

The only information about this combination on google are my own
unanswered questions.