arduino/Arduino #9615: SERCOM UART tutorial shortcomings

jandrassy, 2020-01-12:
in https://www.arduino.cc/en/Tutorial/SamdSercom

in the UART tutorial the RX and TX pins are swapped

Uart mySerial (&sercom3, 0, 1, SERCOM_RX_PAD_1, UART_TX_PAD_0)

for sercom3 pad 1 is pin 1 so RX is pin 1 (SERCOM_RX_PAD_1 = this sercom is uart with RX on pad 1)
for sercom3 pad 0 is pin 0 so TX is pin 0 (UART_TX_PAD_0 = uart TX pad is pad 0)

the first two parameters of the constructor should be 1, 0 but the constructor doesn't care. it only sets the default pin modes for the pins (which works for Serial1 but not here since pin 1 and 0 are io pins at default)

then in setup() the right pin modes are set for the pins 0 and 1 but the comments are wrong. it should be

  pinPeripheral(0, PIO_SERCOM); //Assign SERCOM function to pin 0 (TX)
  pinPeripheral(1, PIO_SERCOM); //Assign SERCOM function to pin 1 (RX)

closed 2020-10-29 by dcuartielles

jandrassy, 2020-10-29:
it still says 0 is RX and 1 is TX

per1234, 2020-10-29:
Hi @jandrassy. Thanks for your report, and also for checking up on it!

The fix has been made in the documentation's source content. This issue was linked to that PR. So when the PR was merged, the issue was closed automatically. The change still needs to be deployed to the Arduino website before it will be publicly visible. I'll give you a preview of the changes so you can review and let me know if any additional changes are needed:
```diff
diff --git a/content/en/Tutorial/SamdSercom/SamdSercom.md b/content/en/Tutorial/SamdSercom/SamdSercom.md
index a252794..c00b418 100644
--- a/content/en/Tutorial/SamdSercom/SamdSercom.md
+++ b/content/en/Tutorial/SamdSercom/SamdSercom.md
@@ -306,7 +306,7 @@ is used to allow the real I2C communication, since the Wire library relies on in
 ```c
 /*
 
-  AnalogReadSerial on new UART placed on pins 0 and 1
+  AnalogReadSerial on new UART placed on pins 1 and 0
 
   Reads an analog input on pin A0, prints the result to the serial monitor.
 
@@ -314,7 +314,7 @@ is used to allow the real I2C communication, since the Wire library relies on in
 
   Attach the center pin of a potentiometer to pin A0, and the outside pins to +3.3V and ground.
 
-  Short together pin 0 and pin 1 with a wire jumper
+  Short together pin 1 and pin 0 with a wire jumper
 
  Created 20 Jun 2016
 
@@ -329,7 +329,7 @@ is used to allow the real I2C communication, since the Wire library relies on in
 #include <Arduino.h>
 #include "wiring_private.h"
 
-Uart mySerial (&sercom3, 0, 1, SERCOM_RX_PAD_1, UART_TX_PAD_0); // Create the new UART instance assigning it to pin 0 and 1
+Uart mySerial (&sercom3, 1, 0, SERCOM_RX_PAD_1, UART_TX_PAD_0); // Create the new UART instance assigning it to pin 1 and 0
 
 // the setup routine runs once when you press reset:
 void setup() {
@@ -340,9 +340,9 @@ void setup() {
 
   mySerial.begin(9600);
 
-  pinPeripheral(0, PIO_SERCOM); //Assign RX function to pin 0
+  pinPeripheral(1, PIO_SERCOM); //Assign RX function to pin 1
 
-  pinPeripheral(1, PIO_SERCOM); //Assign TX function to pin 1
+  pinPeripheral(0, PIO_SERCOM); //Assign TX function to pin 0
 }
 
 // the loop routine runs over and over again forever:
@@ -378,9 +378,9 @@ void SERCOM3_Handler()
 as we did for Wire, the two instructions
 
 ```c
-pinPeripheral(0, PIO_SERCOM); //Assign RX function to pin 0
+pinPeripheral(1, PIO_SERCOM); //Assign RX function to pin 1
 
-pinPeripheral(1, PIO_SERCOM); //Assign TX function to pin 1
+pinPeripheral(0, PIO_SERCOM); //Assign TX function to pin 0
 ```