arduino/Arduino #9616: Detect external changes

evaarnas, 2020-01-13:
In the Arduino IDE it would be useful, if it would detect external change(s) to the file(s) open for editing. Then the user could select whether to save the contents of the editor or reload from disk, or even merge the contents.

per1234, 2020-01-13:
Closing as duplicate of https://github.com/arduino/Arduino/issues/4551

This feature is already in the works (though without the confirmation you suggested) here: https://github.com/arduino/Arduino/pull/6074

closed 2020-01-13 by per1234

per1234, 2020-01-13:
FYI, this feature is already available in the beta build of the Arduino IDE if you want to try it out:
https://www.arduino.cc/en/Main/Software#beta

evaarnas, 2020-01-13:
It seems to work, unless I have _unsaved_ changes in the Arduino IDE. That
is, for "clean" files (all changes saved) it detects the external changes,
but not for "dirty" files (with unsaved changes in the Arduino IDE).