arduino/Arduino #9617: Error in downloading on arduino IDE

Eddylath, 2020-01-13:
I am using Arduino 1.8.10.
I added http://arduino.esp8266.com/stable/package_esp8266com_index.json to Boards Manager URLs and got a downloading error.

"Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json" 
is the error showing.
I have tried deleting sam folder, deleting .json files in hardware folder but nothing worked.
what should i do?

per1234, 2020-01-22:
This error could be caused by a temporary outage of http://arduino.esp8266.com/stable/package_esp8266com_index.json. Does the error still occur @Eddylath?

Eddylath, 2020-01-23:
yes @per1234 
Even i cant download or update anything there.
Why is it occurring?

cmaglie, 2020-01-24:
can you open http://arduino.esp8266.com/stable/package_esp8266com_index.json from a browser? it may be that some network setting? (proxy or whatever)?

Eddylath, 2020-01-26:
yes, I have downloaded it from mozilla a lot of times. It worked perfectly

ubdussamad, 2020-03-27:
HI @Eddylath  , are you using a proxy or something?

MisterAwesome23, 2020-04-10:
Might sound off, but can you try "https" instead of "http"
I'm using the same (with https) and not able to reproduce the error.
Just try this out once and update here.