arduino/Arduino #9618: [Ubuntu] IDE not displaying spaces, tabs and digits

git-yoyo, 2020-01-13:
Installed Arduino IDE version **1.8.10** 64bit on Ubuntu 19.10. The software seems to be working fine but digits, spaces and tabs do not display when writing/pasting/opening code in the editor, which makes it completely unusable. 
(I can delete tabs/spaces/digits and they display correctly when copy-pasting to a text file, so it's probably just a display problem).
Same issue with the console below as you can see in this screenshot:

[image]

facchinm, 2020-01-14:
Hi @git-yoyo ,
which is the default locale in your machine? Did you install any non standard font? Thanks for reporting!

git-yoyo, 2020-01-14:
Hi @facchinm ,

System language is set to English (United States) and input to French, here is the full /etc/default/locale file:

LANG="en_US.UTF-8"
LC_NUMERIC="fr_FR.UTF-8"
LC_TIME="fr_FR.UTF-8"
LC_MONETARY="fr_FR.UTF-8"
LC_PAPER="fr_FR.UTF-8"
LC_NAME="fr_FR.UTF-8"
LC_ADDRESS="fr_FR.UTF-8"
LC_TELEPHONE="fr_FR.UTF-8"
LC_MEASUREMENT="fr_FR.UTF-8"
LC_IDENTIFICATION="fr_FR.UTF-8"

I did not install any font on my system, though I ran
`sudo apt install libcanberra-gtk-module libcanberra-gtk3-module` after noticing a `Failed to load module "canberra-gtk-module"` error when running arduino from the terminal. It fixed the error but not the display issue, so I don't think it had anything to do with my problem.

ubdussamad, 2020-03-27:
Hi @git-yoyo  , is this issue just localized to the editor or is it happening in the UART ( Serial monitor ) window as well?

bparker667, 2020-12-07:
I am having the same issue... has anyone figured out how to fix this yet?