arduino/Arduino #9619: Chomebook / Arduino Create Link error

mpwteach, 2020-01-13:
As of this morning Jan 13, 2020, all of my students are unable to load sketches onto their Chromebooks.  This was working fine two days ago...

Here is the output from a simple blink program.

/home/builder/opt/arduino-builder/arduino-builder -compile -core-api-version 10611 -hardware /home/builder/opt/arduino-builder/hardware -hardware /home/builder/.arduino15/packages -tools /home/builder/opt/arduino-builder/tools -tools /home/builder/.arduino15/packages -built-in-libraries /home/builder/opt/libraries/latest -logger humantags -fqbn arduino:avr:uno -build-cache /tmp -build-path /tmp/718722975/build -verbose -prefs runtime.tools.CMSIS.path=/home/builder/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs runtime.tools.arduinoOTA.path=/home/builder/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs runtime.tools.arm-none-eabi-gcc.path=/home/builder/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs runtime.tools.arc-elf32.path=/home/builder/.arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1 -prefs runtime.tools.arduino101load.path=/home/builder/.arduino15/packages/Intel/tools/arduino101load/2.0.1 -prefs runtime.tools.xtensa-lx106-elf-gcc.path=/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -prefs runtime.tools.bossac.path=/home/builder/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs runtime.tools.x86-linux-gcc.path=/home/builder/.arduino15/packages/arduino/tools/x86-linux-gcc/7.2.0 -prefs runtime.tools.avr-gcc.path=/home/builder/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs runtime.tools.flashpack.path=/home/builder/.arduino15/packages/Intel/tools/flashpack/2.0.0 -prefs runtime.tools.i586-poky-linux-uclibc.path=/home/builder/.arduino15/packages/Intel/tools/i586-poky-linux-uclibc/1.6.2+1.0 -prefs runtime.tools.esptool.path=/home/builder/.arduino15/packages/esp8266/tools/esptool/2.5.0-3-20ed2b9 -prefs runtime.tools.dfu-util.path=/home/builder/.arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1 -prefs runtime.tools.CMSIS-Atmel.path=/home/builder/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs runtime.tools.avrdude.path=/home/builder/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs runtime.tools.core2-32-poky-linux.path=/home/builder/.arduino15/packages/Intel/tools/core2-32-poky-linux/1.6.2+1.0 -prefs runtime.tools.sketchUploader.path=/home/builder/.arduino15/packages/Intel/tools/sketchUploader/1.6.2+1.0 -prefs runtime.tools.mkspiffs.path=/home/builder/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs runtime.tools.linuxuploader.path=/home/builder/.arduino15/packages/arduino/tools/linuxuploader/1.5.1 -prefs runtime.tools.arm-linux-gcc.path=/home/builder/.arduino15/packages/arduino/tools/arm-linux-gcc/4.9.3 -prefs runtime.tools.openocd.path=/home/builder/.arduino15/packages/Intel/tools/openocd/0.9.0+0.1 -prefs runtime.tools.nrf5x-cl-tools.path=/home/builder/.arduino15/packages/arduino/tools/nrf5x-cl-tools/9.3.1 -libraries /tmp/718722975/custom -libraries /tmp/718722975/pinned /tmp/718722975/1blink

Linking everything together...

lto1: fatal error: bytecode stream in file '/tmp/718722975/build/../../core/core_arduino_avr_uno_0fd389f0184d5a21fcc443ff311da7f1.a' generated with LTO version 4.1 instead of the expected 6.0

compilation terminated.

lto-wrapper: fatal error: /home/builder/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-gcc returned 1 exit status

compilation terminated.

/home/builder/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: error: lto-wrapper failed

collect2: error: ld returned 1 exit status

exit status 1

kflanaryolayvar, 2020-01-14:
I just wanted to boost this. My whole class also got the same error. Everything worked fine last week.

per1234, 2020-01-14:
According to https://github.com/arduino/arduino-create/issues/1#issuecomment-351344954, the [Create > Editor section of the Arduino Forum](http://forum.arduino.cc/index.php?board=101.0) is the appropriate place to submit bug reports or feature requests related to the Arduino Web Editor.

So I'm going to close this issue, but there are several reports on the forum and I have already notified the Arduino Web Editor developers of this issue earlier today.

closed 2020-01-14 by per1234