arduino/Arduino #9621: [Library Manager] Add new library

jaenrig-ifx, 2020-01-14:
Could you please add this library to the registrar? 
https://github.com/Infineon/TLE493D-3DMagnetic-Sensor
Thanks!

per1234, 2020-01-14:
Hi @jaenrig-ifx. Libraries added to the Library Manager index are prohibited from containing .exe files. I see your library contains .exe files in the DrillTriger and RotatableJoystick examples, and thus can't be added.

If you still want to have your library added to the Library Manager index, you'll need to remove these files (perhaps add them as assets on the Releases page, or in a different repository). After doing so, please either create a new tag/release (after updating the `version` value in library.properties), or else redo the existing tag/release. Then comment here to let us know we can proceed.

If you no longer want to add your library to the Library Manager index, you can close the issue.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

jaenrig-ifx, 2020-01-14:
Hi @per1234! Thanks for the clarification :) I will be back after taking care of the changes 👍 For now, I will just close the issue. 
Thanks!

jaenrig-ifx, 2020-01-21:
Hi @per1234! Moved the .exe to assets :) If I didn't miss anything. It should be now compliant. Thanks!

jaenrig-ifx, 2020-01-21:
I forgot... I guess I have to delete the previous release then since the .exe files are present in that one. Is it okay if it remains as tag?

per1234, 2020-01-21:
I'm not 100% sure (cmaglie will let us know if I'm wrong), but you should be able to leave the tags and releases that had the .exe file. Those releases won't be available for installation via Library Manager because it only picks up the tags that meet all the requirements, but they shouldn't cause any problems. We just need one of the tags to be compliant. I have checked your `V1.1.1` tag and it's fine, so I think we are all set.

cmaglie, 2020-01-28:
Done.

closed 2020-01-28 by cmaglie

jaenrig-ifx, 2020-02-04:
Thanks @cmaglie !