arduino/Arduino #9624: [Board Manager] 3rd Party Board are not showing in Board Manager

mluthfialhadi, 2020-01-15:
Hi,
I want to add my own board to the Board Manager.
I have make the json file and put it in "Additional Board Manager URLs".
But I can not see my board in the Board Manager.
How can I make it shown in the Board Manager ?

this is my page link about my board at github:
https://github.com/mluthfialhadi/mluthfialhadi.github.io/tree/Rainboard-Board-Manager/Rainboard

(sorry for my bad English)

per1234, 2020-01-15:
Hi @mluthfialhadi. The reason is that your file name doesn't meet the standard:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification#naming-of-the-json-index-file
>The file must be named as follows:
>
>`package_YOURNAME_PACKAGENAME_index.json`
>
>The prefix package_ and the postfix _index.json are mandatory (otherwise the index file is not recognised by the IDE)

I'm happy to provide further assistance with your project, but let's use the [Arduino Forum](https://forum.arduino.cc/) for that, as this tracker is meant to be used only for bug reports and feature requests.

closed 2020-01-15 by per1234

mluthfialhadi, 2020-01-15:
Thank you very much