arduino/Arduino #9628: Windows app not working correctly

bee9e, 2020-01-15:
I downloaded the windows app for arduino ide. But i got constant compiling error that did not make sense. I then uninstalled the app and downloaded the original Ide. I then used the exact same code that got compiling errors. And it compiled and uploaded with no problems.
I think it maybe has something to do with the the way libraries are handled by the Ide.

per1234, 2020-01-22:
I @bee9e in order to investigate this, we'd need you to post the complete text of the error output. Please re-install the Windows app, reproduce the error, then post it in a reply here.

per1234, 2020-06-14:
Closing due to lack of response.

@bee9e if you're still having this issue and will provide the requested information I'll be happy to re-open this issue.

closed 2020-06-14 by per1234