arduino/Arduino #9633: IDE serial monitor minimum width

torsj, 2020-01-17:
The size of the IDE serial monitor window is adjustable by grabbing the edge of the window, but only down to a minimum.

The minimum width of the IDE serial monitor window is too large and it fills a large portion of the screen.
This behavior is new with 1.8.10 (or possibly earlier). I use mac os. The limits to the size of the window should be similar to the limits for the main IDE window.

The screen shot shows both windows at their minimum size.

[image]

tistructor, 2020-02-06:
The same problem occurs on arduino 1.8.11.

The serial monitor does not resize horizontally, not allowing at least on linux mint to automatically resize the window by dragging it to the edge of the screen so as to perfectly support it with the editor.
Instead, the plot resizes.

I tried to solve this problem by changing the parameters in preferences.txt but failed.

Initially I had the same problem with the editor window but then by changing the size in the preferences.txt file it was fixed because I did not know the fact that if the minimum horizontal size of the window is more than half of the screen this does not resize with the method of dragging it to the edge of the descktop.

I believe there is a fixed value somewhere in the application code.

tistructor, 2020-02-16:
The same problem occurs on arduino 1.8.12

torsj, 2020-07-07:
The problem is still there on 1.8.13

leosdad, 2021-07-31:
Quite annoying, eswpecially if you need multiple instances. It's an easy fix, any news on this issue?

Louis-DR, 2025-02-05:
It's been many years. On MacBook, the serial monitor takes over half of the width of the screen, making it impractical to have both the code and the serial monitor at once. Is there a fix planned ?