arduino/Arduino #9635: [Library Manager] Add the following Library

nitin710, 2020-01-17:
Please add the following Library to the Arduino Library Manager:
1. https://github.com/EmotiBit/EmotiBit_XPlat_Utils

per1234, 2020-01-18:
Hi @nitin710. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

nitin710, 2020-01-18:
Created the release.

nitin710, 2020-01-21:
As requested, i have created a release for this library.
When can i expect this library to be added  to the Arduino Library Manager?

per1234, 2020-01-22:
cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

cmaglie, 2020-01-28:
Done.

closed 2020-01-28 by cmaglie