arduino/Arduino #9651: "undefined reference to operator delete(void*, unsigned int)"

Deema35, 2020-01-21:
Hi. 
When I try to compile my project (https://github.com/Deema35/Arduino_GPS_Clock) on Arduino IDE 1.8.10. I am get Error "undefined reference to operator delete(void*, unsigned int)". Arduino IDE refers to file (https://github.com/Deema35/Arduino_GPS_Clock/blob/master/GPS_ClockState.cpp). If I am down grade Arduino IDE to 1.8.8 (1.8.9 not try) project will compile without Errors. Thank you.

Deema35, 2020-01-21:
Addition. I have try 1.8.9 and it too compile without Errors.

cmaglie, 2020-01-24:
Please enable "Verbose compiler" from preferences, try to build, and copy here the full output.
Also knowing your OS and board may help.

Deema35, 2020-01-24:
I am use OS Windows and board WavGat Uno R3 CH340G(chinese clone). Board file: https://drive.google.com/file/d/10gwrG9uTDwaEO-7EudsmBkfgdcyrcABI/view . 
My output file in attacment.

P.S. I am try to compile with origin UNO and code compile without Errors.
[Arduino_WavGat.txt](https://github.com/arduino/Arduino/files/4108296/Arduino_WavGat.txt)

matthijskooijman, 2020-01-29:
It seems you are using the "WAVGAT UNO r3" board. A quick google shows the installation instructions at https://www.pcboard.ca/arduino-uno-r3-wavgat-setup, which shows that this board uses its own Arduino core (i.e. the code that is compiled onto the Arduino) instead of the core supplied by Arduino. This is also confirmed by the "board file" you linked. This core is also the place where such a delete function should be defined, so it is likely your problem should be solved in the WAVGAT uno core, rather than in the Arduino IDE.

I suggest you find out where to report bugs or request support for this WAVGAT uno core, and ask there.

Since I suspect this is not an Arduino IDE or official Arduino core bug, I'm closing this issue. If closer inspection reveals that there is an IDE bug involved, feel free to comment and we can reopen this issue.

closed 2020-01-29 by matthijskooijman