arduino/Arduino #9653: analogWrite - PWM Problem in IDE-Version 1.8.10

Nnambles, 2020-01-22:
There is a Bug in Version 1.8.10 of the Arduino IDE using analogWrite() for PWM on Arduino Mega 2560. If i use analogWrite() with any value for the pwm-Pins once in the loop()-Method everything works fine. If I put a second call of analogWrite() with the same pin and any value that is not 0 or 255, nothing happens. The loop-Method keeps on working, but the PWM-Pin used in analogWrite() shows an continous output of 0V. In Version 1.8.9 the same programm works fine.

alaqadi, 2020-02-01:
I discovered the same issue today as I teach Arduino to my students , and a few of them has the issue  of when putting a second analogWrite() for a second PWM on the same pin or a different pin, it will case online one PWM signal to appear. This happened on Mega 2560. After testing multiple boards with multiple Oscilliscopes, I was finally to find the reason as I tested  one board on a computer that had 1.8.8 and  it worked fine. I tried 1.8.11 it had the same problem.
Please fix or remove 1.8.10 and 1.8.11 from the main download page. This is a major bug

facchinm, 2020-02-17:
Should be moved to ArduinoCore-AVR and fixed ASAP (although I think it's due to the compiler update more than any core update)

Rocketct, 2020-02-25:
Hi all, i made some tests, and i found that:

1. Seems taht as @facchinm  sayas the issue does not depend from the core AVR but from the compiler;
2.  setting, for the analogWrite, the attribute  `__attribute__((optimize(**...**)));` and giving some properly value to the optimize it is possible make it works properly, i get the following results:

-  `__attribute__((optimize("-O0")));  `
WORKS:
The sketch use:
**flash**: 2332 byte
**ram**: 9 byte

-  `__attribute__((optimize("-01")));  `
WORKS:
Sketch size:
**flash** 1964 byte
**ram**: 9 byte

- `__attribute__((optimize("-02"))); `
WORKS
 Sketch size:
**flash**: 1830 byte
**ram**: 9 byte

- `__attribute__((optimize("-0s"))); `
NOT  WORKS

Sketch size:
**flash**: 1830 byte
**ram**: 9 byte

- Without any `__attribute__ `
NOT  WORKS
Sketch size:
**flash**: 1836 byte
**ram**: 9 byte

the AVR- Core patch to test it is the following:
```
diff --git a/cores/arduino/Arduino.h b/cores/arduino/Arduino.h
index 91eeb16..b86496f 100644
--- a/cores/arduino/Arduino.h
+++ b/cores/arduino/Arduino.h
@@ -136,7 +136,7 @@ void digitalWrite(uint8_t pin, uint8_t val);
 int digitalRead(uint8_t pin);
 int analogRead(uint8_t pin);
 void analogReference(uint8_t mode);
-void analogWrite(uint8_t pin, int val);
+void analogWrite(uint8_t pin, int val) __attribute__((optimize("-O0"))); unsigned long millis(void);
 unsigned long micros(void);
```

matthijskooijman, 2020-02-25:
Can anyone post a short example sketch that shows the problem, along with a description of what (incorrect) output it shows?

Rocketct, 2020-02-25:
yes sure, simple:

```
void writeTest(int a) {
  analogWrite(3, a);
  delay(5000);
}

void setup() {
}

void loop()
{
  writeTest(128);
  writeTest(64);
}
```

matthijskooijman, 2020-02-25:
@Rocketct  And can you also describe the output you get when you run this particular sketch?

Rocketct, 2020-02-25:
yes sorry added on the previous comment let me say if  clear :smile:

matthijskooijman, 2020-02-25:
@Rocketct, Thanks, perfectly clear. I have tried your sketch and looked at the generated machine code, which actually looks to be correct. I don't have an Arduino at hand right now to check if the same problem occurs for me, though.

What Arduino did you test this with?

matthijskooijman, 2020-02-25:
Ah, confirmed here: Uno works, Mega is broken. Gotta run now, I hope to look at the assembly again in the next days.

matthijskooijman, 2020-02-25:
I had a look at the generated assembly. Most is similar, except that the analogWrite has a jump table rather than nested branches to select between the different timers (probably because the mega has more timer pins). I originally thought the jump table handling was messed up (wrong RAMPZ value), but while annotating the assembly below, I realized it did seem correct after all. In any case, below is the assembly (comments with `#` added by me):
```
0000031e <analogWrite.part.0.constprop.2>:
# Note that the pin number is constant propagated, only the val is passed in r24
void analogWrite(uint8_t pin, int val)
 31e:   9c 01           movw    r18, r24
                switch(digitalPinToTimer(pin))
# This loads the timer value from a progmem table (item inside the table is at 0x01A7)
 320:   e7 ea           ldi     r30, 0xA7       ; 167
 322:   f1 e0           ldi     r31, 0x01       ; 1
 324:   e4 91           lpm     r30, Z
 # This checks for NOT_A_TIMER or similar, and jumps to 0x48c which handles that case
 326:   e1 50           subi    r30, 0x01       ; 1
 328:   e2 31           cpi     r30, 0x12       ; 18
 32a:   08 f0           brcs    .+2             ; 0x32e <analogWrite.part.0.constprop.2+0x10>
 32c:   af c0           rjmp    .+350           ; 0x48c <__LOCK_REGION_LENGTH__+0x8c>
# This is the jump table part. It takes the value returned by digitalPinToTimer(pin)
# (r30), extends it to r30:r31:r24 (little-endian, and adds 0x19c (by
# subtracting 0xfffe64, which wraps
# around to add 0x19c, which is half of 0x338 which is where the jump table
# starts). tablejump2 below then doubles this value, so it ends up looking
# at 0x338 + 2*digitalPinToTimer(pin), which makes sense since the jump table
# starts at 0x338 and each entry is 2 bytes (note that a timer 0 is invalid, so
# the table actually starts at 0x338 + 2).
 32e:   f0 e0           ldi     r31, 0x00       ; 0
 330:   88 27           eor     r24, r24
 332:   e4 56           subi    r30, 0x64       ; 100
 334:   fe 4f           sbci    r31, 0xFE       ; 254
 336:   8f 4f           sbci    r24, 0xFF       ; 255
 338:   ef c1           rjmp    .+990           ; 0x718 <__tablejump2__>
# This is the jump table, the first entry for digitalPinToTimer(pin) == 1
# This jumps to 0x01af (word address, so byte address 0x35e) which seems sane
 33a:   af 01           movw    r20, r30
 33c:   b4 01           movw    r22, r8
 33e:   b9 01           movw    r22, r18
 340:   c3 01           movw    r24, r6
 342:   cd 01           movw    r24, r26
 344:   46 02           muls    r20, r22
 346:   d7 01           movw    r26, r14
 348:   df 01           movw    r26, r30
 34a:   e7 01           movw    r28, r14
 34c:   f1 01           movw    r30, r2
 34e:   fb 01           movw    r30, r22
 350:   05 02           muls    r16, r21
 352:   14 02           muls    r17, r20
 354:   1e 02           muls    r17, r30
 356:   46 02           muls    r20, r22
 358:   28 02           muls    r18, r24
 35a:   32 02           muls    r19, r18
 35c:   3c 02           muls    r19, r28

# The jump table ends here, and this is the first case handler
                        case TIMER0A:
                                // connect pwm to pin on timer 0, channel A
                                sbi(TCCR0A, COM0A1);
 35e:   84 b5           in      r24, 0x24       ; 36
 360:   80 68           ori     r24, 0x80       ; 128
                                OCR0A = val; // set pwm duty
 362:   84 bd           out     0x24, r24       ; 36
 364:   27 bd           out     0x27, r18       ; 39
 366:   08 95           ret
# This function continues with more case handlers
...

00000718 <__tablejump2__>:
 718:   ee 0f           add     r30, r30
 71a:   ff 1f           adc     r31, r31
 71c:   88 1f           adc     r24, r24
# I suspect this is the RAMPZ register, which contains the upper bits for a
# > 16bits elpm (should be 0 here)
 71e:   8b bf           out     0x3b, r24       ; 59
 720:   07 90           elpm    r0, Z+
 722:   f6 91           elpm    r31, Z
 724:   e0 2d           mov     r30, r0
 726:   19 94           eijmp
```

I guess this either needs some debug prints (to read back various register values) or a JTAG debugger to verify what parts are correct and where the problem starts (or looking at the asm more closely and really verifying that all the values are correct). I probably won't have more time for this soon, though.

matthijskooijman, 2020-02-26:
Also, it might be useful to run with `-fno-jump-tables` to see if that also prevents the problem.

Rocketct, 2020-02-26:
okk @matthijskooijman @PaulStoffregen thank you, in meanwhile will discuss with @facchinm about next steps

Rocketct, 2020-02-26:
@matthijskooijman tested with `-fno-jump-tables` and works properly

Potul72, 2020-03-05:
I have tested in IDE version 1.8.7, 1.8.9 and 1.8.12 and happens to all versions, so I do not understand how @alaqadi and @Nnambles could make it run in previous versions.

`void setup() {
  // put your setup code here, to run once:
analogWrite (7, 100);
analogWrite (8, 200);

}

void loop() {
  // put your main code here, to run repeatedly:

}`  This is what I am testing and with the oscilloscope I only see signal at pin 7...
[image]

matthijskooijman, 2020-03-05:
@Potul72 Did you check the AVR core version used? If you upgraded that through the boards manager, a newer version is downloaded into your user folder, which then replaces the IDE-bundled version, even when you change the IDE version I think. However, if you do not have the AVR core in your user folder, it will use the IDE bundled version which will then change with IDE changes (usually).

To check, you can enable verbose compilation in the preferences, it should then use the AVR core version logged near the top of the compilation output.

Using the board manager, you can also easily test different AVR core versions without having to change the IDE version. You can revert to the bundled version, I think, by selecting the bundled version number in the board manager and "installing" it (I think this just throws away the version from the user folder, reverting to the IDE bundled version again).

Potul72, 2020-03-06:
Thanks @matthijskooijman. I have tested IDE version 1.8.5 with AVR core 1.8.2 and it failed also, then I tested IDE 1.8.5. with AVR core 1.6.12 and It work wounderfull. Then I tested latest IDE version 1.8.12 with AVR core 1.6.23 and it did work! So bug is on AVR Boards core from version 1.8.1, as there are no downloable version beween 1.6.23 and 1.8.1.

WestfW, 2020-03-07:
I did a bit of tracing in the simulator.  It looks like the code is ending up in the case for TIMER3B, when pin3 OUGHT to be TIMER3C - some sort off-by-one error in doing the jump table stuff?
Here's a dump of some of the code, and just before the jump to __tablejump2__, it looks like the "subtracted" constant for the "new" code is before the actual table?  Some of the surrounding code has changed significantly, so it's hard to tell.

**New:**
```
// Right now, PWM output only works on the pins with
// hardware support.  These are defined in the appropriate
// pins_*.c file.  For the rest of the pins, we default
// to digital output.
void analogWrite(uint8_t pin, int val)
 31e:   9c 01           movw    r18, r24
        {
                digitalWrite(pin, HIGH);
        }
        else
        {
                switch(digitalPinToTimer(pin))
 320:   e7 ea           ldi     r30, 0xA7       ; 167
 322:   f1 e0           ldi     r31, 0x01       ; 1
 324:   e4 91           lpm     r30, Z
 326:   e1 50           subi    r30, 0x01       ; 1
 328:   e2 31           cpi     r30, 0x12       ; 18
 32a:   08 f0           brcs    .+2             ; 0x32e <analogWrite.part.0.constprop.2+0x10>
 32c:   af c0           rjmp    .+350           ; 0x48c <__LOCK_REGION_LENGTH__+0x8c>
 32e:   f0 e0           ldi     r31, 0x00       ; 0
 330:   88 27           eor     r24, r24
 332:   e4 56           subi    r30, 0x64       ; 100  24bits of (-(0x338 >> 1)), one word before jump table (?)
 334:   fe 4f           sbci    r31, 0xFE       ; 254     :
 336:   8f 4f           sbci    r24, 0xFF       ; 255       :
 338:   94 c1           rjmp    .+808           ; 0x662 <__tablejump2__>
 
 33a:   af 01           movw    r20, r30	; (case table) timer0a
 33c:   b4 01           movw    r22, r8
 33e:   b9 01           movw    r22, r18
 340:   c3 01           movw    r24, r6
 342:   cd 01           movw    r24, r26
 344:   46 02           muls    r20, r22
 346:   d7 01           movw    r26, r14
 348:   df 01           movw    r26, r30
 34a:   e7 01           movw    r28, r14
 34c:   f1 01           movw    r30, r2
 34e:   fb 01           movw    r30, r22
 350:   05 02           muls    r16, r21
 352:   14 02           muls    r17, r20

000001a4 <digital_pin_to_timer_PGM>:
 1a7:   02 09 0c 0d 0e 08 07 03 04 01 00 00     ................
```

**Old:**
```
        {
                digitalWrite(pin, HIGH);
 392:   81 e0           ldi     r24, 0x01       ; 1
 394:   61 cf           rjmp    .-318           ; 0x258 <digitalWrite.constprop.
2>
        }
        else
        {
                switch(digitalPinToTimer(pin))
 396:   ef ee           ldi     r30, 0xEF       ; 239
 398:   f1 e0           ldi     r31, 0x01       ; 1
 39a:   e4 91           lpm     r30, Z
 39c:   8e 2f           mov     r24, r30
 39e:   90 e0           ldi     r25, 0x00       ; 0
 3a0:   fc 01           movw    r30, r24
 3a2:   31 97           sbiw    r30, 0x01       ; 1
 3a4:   e2 31           cpi     r30, 0x12       ; 18
 3a6:   f1 05           cpc     r31, r1
 3a8:   08 f0           brcs    .+2             ; 0x3ac <analogWrite.constprop.0
+0x62>
 3aa:   9c c0           rjmp    .+312           ; 0x4e4 <__LOCK_REGION_LENGTH__+
0xe4>
 3ac:   88 27           eor     r24, r24
 3ae:   ec 57           subi    r30, 0x7C       ; 124 24bits of (-(0x108 >> 1)), one word before jump table (?)
 3b0:   ff 4f           sbci    r31, 0xFF       ; 255   :
 3b2:   8f 4f           sbci    r24, 0xFF       ; 255     :
 3b4:   63 c1           rjmp    .+710           ; 0x67c <__tablejump2__>

; timer table
000001ec <digital_pin_to_timer_PGM>:
 1ef:   02 09 0c 0d 0e 08 07 03 04 01 00 00

; case table earlier in memory:
 108:   db 01           movw    r26, r22  ;timer0a case
 10a:   e0 01           movw    r28, r0	  ;timer0b case
 10c:   e5 01           movw    r28, r10  ;timer1a case
 10e:   ef 01           movw    r28, r30  ;timer1b case
 110:   f9 01           movw    r30, r18  ;timer1c case
 112:   72 02           muls    r23, r18
 114:   03 02           muls    r16, r19
 116:   0b 02           muls    r16, r27
 118:   13 02           muls    r17, r19
 11a:   1d 02           muls    r17, r29
 11c:   27 02           muls    r18, r23
 11e:   31 02           muls    r19, r17
 120:   40 02           muls    r20, r16
 122:   4a 02           muls    r20, r26
 124:   72 02           muls    r23, r18
 126:   54 02           muls    r21, r20
 128:   5e 02           muls    r21, r30
 12a:   68 02           muls    r22, r24
```

igendel, 2020-03-08:
Hello, excuse me for the interruption - I stumbled upon this issue lately and found this thread... if it helps in any way to pinpoint the source of the problem, I noticed that when I pass the duty cycle parameter to analogWrite using a variable declared as **volatile**, the problem disappears. Probably related to the optimization settings @Rocketct mentioned above.

victor045, 2020-03-11:
Hello, I can confirm that the issue happened to me with the IDE versions 1.8.10, and 1.8.13.
It worked with the previous 1.8.8 version.

elilitko, 2020-03-24:
The issue is definitely confirmed for IDEs 1.8.10 and higher. Just try to run this simple sketch on Mega2560

`void setup(void) {
	analogWrite(6, 100);
	analogWrite(5, 200);
}

void loop(void) {}`

and make sure - there is no PWM on those pins.

**### The workaround is found**!

Open the file platforms.txt and remove all the options concerned to lto. Just look for "lto" with context search and remove all of them. 

That's all. Have a nice day!

Potul72, 2020-04-29:
Thanks @elilitko! just let me clarify for other users (beginers like me) to remove only the options -lto and -flto. ... I removed the complete line ... and only the one containing -lto ...

mark-berobot, 2020-05-02:
Confirmed this issue with Windows 10, Arduino 1.8.12 (Windows Store 1.8.33.0)

https://arduino.stackexchange.com/questions/75338/mega-2560-simple-program-one-analogwrite-turns-off-the-other

matthijskooijman, 2020-05-03:
I think this is a problem that we should somehow tackle. How about:
 - Adding `-fno-jump-tables` to the compilation options to prevent it from happening? This can hurt performance of big switch tables, but should otherwise not affect sketch performance much I think, so I would say this is acceptable for now. @facchinm, @PaulStoffregen, what would you think?
 - Trying to make a minimized example out of this, so we can report a bug at gcc upstream? With minimized, I do not mean a small sketch, but plain .c / .cpp files that can be fed directly to gcc without needing any Arduino code. Anyone want to have a stab at this? I won't have time in near future...
 - Moving this issue to ArduinoCore-avr?

facchinm, 2020-05-04:
1. I think it's a good workaround we can implement immediately 
2. gcc upstream status is a bit cumbersome now (since they called for removal of the AVR backend)
3. absolutely, doing it now :wink: