arduino/Arduino #9655: Arduino IDE 1.8,10 - Problem with own Servo library on ESP32

JarekParal, 2020-01-22:
Hello, 

My library [ServoESP32](https://github.com/RoboticsBrno/ESP32-Arduino-Servo-Library) isn't compilable with the latest version Arduino IDE 1.8.10. 
I created this library 3 years ago to add support for controlling servos on ESP32 with the same API as the original Arduino servo library (which wasn't available on ESP32 and is still not available).

The library has the same name of `.h/.cpp` files like the original Arduino Servo library:
- `Servo.h`
- `Servo.cpp`

But until version 1.8.10 there wasn't any problem.

In the version 1.8.9 everything work correctly:
```sh
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\noname\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\noname\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\noname\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\noname\AppData\Local\Temp\arduino_build_517251 -warnings=all -build-cache C:\Users\noname\AppData\Local\Temp\arduino_cache_369272 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\noname\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library\examples\02-ServoPotentiometer\02-ServoPotentiometer.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\noname\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\noname\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\noname\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\noname\AppData\Local\Temp\arduino_build_517251 -warnings=all -build-cache C:\Users\noname\AppData\Local\Temp\arduino_cache_369272 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\noname\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\noname\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library\examples\02-ServoPotentiometer\02-ServoPotentiometer.ino
Using board 'esp32' from platform in folder: C:\Users\noname\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: C:\Users\noname\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
WARNING: Spurious .pio folder in 'ServoESP32' library
Detecting libraries used...
"C:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\noname\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/ [... cut ...]
Is there some new implementation of library searching module which can cause this issue?

For reproducing this issue you can use one of my examples:
https://github.com/RoboticsBrno/ESP32-Arduino-Servo-Library/blob/master/examples/02-ServoPotentiometer/02-ServoPotentiometer.ino

Thanks for the help.

PS: I don't want to change the names of the files, because that will break compilation for all users of this library after updating the library.

PS2: The build with PlatformIO still works correctly: https://travis-ci.org/RoboticsBrno/ESP32-Arduino-Servo-Library

cmaglie, 2020-01-23:
Thanks for the bug report, I confirm the issue and I'm working on a fix...

JarekParal, 2020-01-23:
@cmaglie Thanks for solving the issue. Do you know when should be released Arduino IDE 1.8.11? The release of the version 1.8.10 was in September 2019.

cmaglie, 2020-01-23:
Yes it will be included in IDE 1.8.11 that is going to be release Monday Jan, 27.

closed 2020-01-23 by cmaglie

cmaglie, 2020-01-24:
A fixed build of the IDE should be available on the hourly build if you want to try it: https://www.arduino.cc/en/Main/Software#hourly