arduino/Arduino #9657: [Library Manager] Please add Audio - Adafruit Fork to the library manager

dherrada, 2020-01-23:
Please add this library to the library manager:
https://github.com/adafruit/Audio

per1234, 2020-01-24:
Hi @dherrada libraries added to the Library Manager index are prohibited from containing .exe files. The adafruit/Audio repository contains one at extras/wav2sketch/wav2sketch.exe.

If you still want the library added to the Library Manager index, you'll need to remove the .exe file from the repository and then either create a new tag/release or redo the existing release. You might consider adding the .exe file as an asset of the GitHub release instead. Once that's done, please comment here so we'll know we can proceed.

If you no longer want to add the library to the Library Manager index, you can go ahead and close this issue.

dherrada, 2020-01-24:
@per1234 I've turned wav2sketch.exe into a zip archive and re-released the library. Does that work?

per1234, 2020-01-24:
I believe so. If I'm wrong, cmaglie will let us know.

cmaglie, 2020-01-28:
The library it's ok now, I've added it.

closed 2020-01-28 by cmaglie