arduino/Arduino #9660: Please update library in library manager

Panjkrc, 2020-01-25:
Please update TCS3200 library to version 1.2.0

Link to github repo: https://github.com/Panjkrc/TCS3200_library
Link to latest release: https://github.com/Panjkrc/TCS3200_library/releases/tag/v1.20-2512020

Thanks in advance!

per1234, 2020-01-25:
Hi @Panjkrc the Arduino Library Manager indexer job runs hourly and automatically picks up all new releases of all libraries in the index, as long as they comply with the [requirements for inclusion in the Arduino Library Manager index](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ). So there is no need to submit a request here every time you make a release. In fact, your 1.2.0 release is already available for installation via Library Manger.

Reference:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

Panjkrc, 2020-01-25:
Ok, thank you!

closed 2020-01-25 by Panjkrc