arduino/Arduino #9669: Arduino Drivers silent install - On Windows XP

marianvoicu2002, 2020-01-27:
Hello,

I use the following lines to install arduino uno drivers
PUSHD .\Drivers
CertUtil -AddStore TrustedPublisher "Certs\CertDrv1.cer"
CertUtil -AddStore TrustedPublisher "Certs\CertDrv2.cer"
CertUtil -AddStore TrustedPublisher "Certs\CertDrv3.cer"
CertUtil -AddStore TrustedPublisher "Certs\CertDrv4.cer"
dpinst-x86.exe /LM /S /SE
POPD
CertUtil -DelStore TrustedPublisher "Adafruit Industries"
CertUtil -DelStore TrustedPublisher "Arduino AG"
CertUtil -DelStore TrustedPublisher "Arduino srl"
CertUtil -DelStore TrustedPublisher "Arduino LLC"

It works for Windows7 X86, but doesn't work for WindowsXp x86.

On Xp works if i remove /S, and install with setup.

Can anyone help me?

TX,
Marian

cmaglie, 2020-02-03:
Perhaps check if the OS if windows XP and execute the dpinst without "/S" in that case?

https://stackoverflow.com/a/13212116/1655275

marianvoicu2002, 2020-02-03:
> Perhaps check if the OS if windows XP and execute the dpinst without "/S" in that case?

If execute without "/S" works but that means we cannot install silent the drivers, isn't it?

cmaglie, 2020-02-03:
ah I see, I misinterpreted your:

> On Xp works if i remove /S, and install with setup.

then no, sorry, I didn't have a working solution for XP, I even don't know if this is possible at this point.

per1234, 2020-10-02:
Since Arduino IDE [is no longer compatible with Windows XP](https://github.com/arduino/Arduino/issues/10018), I'm closing this.

closed 2020-10-02 by per1234