arduino/Arduino #9673: [Library Manager] Please add  Nouryas-Advanced-Line-Follower-Array library

Nouryas-Tech, 2020-01-28:
Hello,

Please add the following library to the Library Manager:
https://github.com/Nouryas-Tech/Nouryas-Advanced-Line-Follower-Array
Thanks!

per1234, 2020-01-28:
Hi @Nouryas-Tech. In order for the library to be added to the Arduino Library Manager index, you will need to do two things:
1. Add a library.properties file to the root of the repository. The file's format is documented here: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
1. Create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After you have done that, please comment here so we'll know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Nouryas-Tech, 2020-01-29:
I hope that I correctly did what you ask for.

Nouryas-Tech, 2020-02-03:
hello , I hope there is nothing is missed which may stop publishing the library.

per1234, 2020-02-03:
@Nouryas-Tech it's fine now. Your library will be added to the Arduino Library Manager index the next time cmaglie comes through to clear all the outstanding requests.

Thanks!

cmaglie, 2020-02-07:
Done.

closed 2020-02-07 by cmaglie