arduino/Arduino #9685: package as .snap for linux

jor1196, 2020-01-30:
Hello
I have been using the packages in snap for a while when installing applications on Linux and recently I came up with the idea of ​​learning to program with these small boards for some future project, now I wanted to download the arduino IDE but the download is a compressed .tar package. xz instead of one of installation, which would be more comfortable since lately these are very easy to use

per1234, 2020-02-04:
There was a previous request for this here: https://github.com/arduino/Arduino/issues/5068

As is mentioned in that issue, there is a 3rd party maintained snap for the Arduino IDE. At one time, people using it would have problems that did not occur when using the official Arduino IDE. I'm not sure whether that issue has been resolved. The snap has not been updated for 1.5 years. This seems to be a common problem with 3rd party maintained packages. It seems to me that the only way this will turn out well is if Arduino maintains an official package.

galgalesh, 2020-05-15:
I recently created a new Arduino IDE snap and deprecated the old one. You can find the packaging files here: https://github.com/snapcrafters/arduino

In less than two months, this snap shot from 0 installs to 13.500 _active_ installations. The Arduino IDE is clearly wanted by Linux users and the snap is one of the easiest ways to install it.

The snap also includes a GUI warning message if the user does not have the required permissions to access Arduino devices over USB. This solves one of the main issues users have when they install Arduino IDE on Linux.

The Snap Store is specifically made so upstream publishers like Arduino can publish packages themselves, to give them more control over the release schedule etc. If you are interested in taking over maintenance of this snap, I'm happy to help you get started!