arduino/Arduino #9693: Arduino IDE version 1.8.11  can't display Thai character.

TridentTD, 2020-01-31:
Arduino IDE version 1.8.10 
can display Thai fine.

But when upgrad to version1.8.11
the lastest version can't display Thai.

[image]

suabmoohoo, 2020-02-02:
The symptoms are the same, so are there any solutions?

TridentTD, 2020-02-03:
> The symptoms are the same, so are there any solutions?

Return to version 1.8.10.

shawayux, 2020-03-02:
@cmaglie, I have been trying to fix the bug, probably an issue with language encoder, can you say me where is the source code of IDE!

shawayux, 2020-03-17:
Hi!
I have figured out why is this happening, The main reason is that the root folder of the Linux, misses mimeapps.list. This can be solved by adding the mimeapp.list to the root folder of the Linux, I will be submitting a PR in 24 hours. By adding mimeapps.list to the root list will also solve the issue  #7846 
As I am just a student and new to open-source, I would request the admin @cmaglie to help me with it, if what I am saying is right. Though I have tested it on PC and it works perfectly fine!

cmaglie, 2020-03-23:
@shawayux I never heard about the `mimeapps.list` file, can you elaborate more? what is the purpose of this file? Could you provide your copy `mimeapps.list` so we can try it?

shawayush, 2020-03-23:
Hi! @cmaglie 
XDG Standard is used for configuring desktop Environment! default application of each MIME type are stored in `mimeapps.list` Generally, it is provided by the distribution of some UNIX, but some OS, unfortunately, doesn't have it, all we have to do is create that file.
I have attached a screenshot of my `mimeapps.list` which was created by the code that I have just given as a PR. I hope you like the work I have done here. Please feel free to guide me during the process.

[image]

cmaglie, 2020-03-24:
@shawayush I don't think your patch solves this particular issue, here users of **Windows** are unable to display strings in Thai language, I don't think that changing the install script for **linux** will solve this problem.

cmaglie, 2020-03-24:
For the record, the problem here is that the IDE 1.8.11 has been distributed with the java runtime OpenJDK (the AdoptOpenJDK build to be precise). Unfortunately OpenJDK does still have a lot of problems with font rendering and 2D graphics, so we had to revert back to Sun/Oracle JRE (the latest version available that is still free redistributable). This fixed the problem for IDE 1.8.12.

BTW we should move to OpenJDK before or later so I'm still keeping this issue open to keep track of the bug. More bugs related to OpenJDK are under the `OpenJDK` label: https://github.com/arduino/Arduino/issues?q=label%3AOpenJDK+

shawayush, 2020-03-24:
@cmaglie does it solve #7846 ?