arduino/Arduino #9694: Error-Message: arduino-builder.exe infected with idp.alexa.51

ProjektHB, 2020-01-31:
Do someone know the problem with the idp.alexa.51 virus? Everytime i want to flash the Arduino uno with a preinstalled example program i get the mentioned virus detection from the avg Software. Can someone help me?

gvarisco, 2020-01-31:
Hi @ProjektHB, thanks for reaching out. Any chance you could provide us the md5/sha1 checksum for the **arduino-builder.exe** file? Happy to follow up with the AVG team.

gvarisco, 2020-01-31:
@ProjektHB can you please let me know which IDE version are you referring to? In the meantime, I've filled a ticket to AVG as it is - most likely - a false positive.

ProjektHB, 2020-01-31:
Its the Version 1.8.10. Thanks for your support.

gvarisco, 2020-01-31:
@ProjektHB I've submitted it as false positive to AVG. The current detection for 1.8.10's arduino-builder.exe is as follows: https://www.virustotal.com/gui/file/c3893284b7ad8b6875e110d3cf2e8391dade3f469d87f850e7c301ef9d3ec41b/detection

AVG's latest definitions file should be solving the issue. Let me know if it still persists in the next 24 hours.

Thank you for the heads up.

gvarisco, 2020-01-31:
AVG just replied:
`Please accept our apologies for the inconvenience caused. The detection by AVG was incorrect and was removed in a recent AVG update, please wait at least 24 hours.
`

Joevlc84, 2020-03-18:
Hi guys,
I also have the same error.
AVG has detected my arduino as containing two virus.

Error in spanish:
Hemos bloqueado **arduino-builder.exe** porque estaba infectado por **IDP.ALEXA.51.**
Hemos bloqueado **arduino-builder.exe** porque estaba infectado por **IDP.Generic.**

My arduino IDE is 1.8.33.0
My AVG is 20.1.2397 (compilación 20.1.5069.559)
Version virus database 200318-0

Regards

gvarisco, 2020-03-19:
Hi @Joevlc84 can you please double-check your IDE's version? Our latest stable is 1.8.12. Looking forward to hearing from you.

Poker-Mike, 2020-03-20:
Hi Gvarisco, got the same issue with Avast at that moment. Very first time trying to push code to the board. Brand new in arduino world, hope it’s safe.

gvarisco, 2020-03-20:
Hi @Poker-Mike thanks for letting us know. I can let you know that is a false positive. We'll ask them to whitelist the binaries once again.

Joevlc84, 2020-03-20:
> 
> 
> Hi @Joevlc84 can you please double-check your IDE's version? Our latest stable is 1.8.12. Looking forward to hearing from you.

Yeah, sorry, as Poker-Mike said

 arduino 1.8.12 (windows store 1.8.33.0)

Thanks :-)

elgybe, 2020-03-27:
Hi,
Same issue with Avast antivirus:
[image]

Hopefully I have the possibility to add an exception.

luigigubello, 2020-04-03:
hI @elgybe, could you share md5/sha1 checksum for **arduino.exe**? So I can contact Avast team, thank you :)

per1234, 2021-04-02:
It sounds like this has been resolved for everyone so I'll go ahead and close it. If anyone is still getting this virus detection from the Arduino IDE, please feel free to comment here and I'll make sure to reopen.

closed 2021-04-02 by per1234