arduino/Arduino #9701: IDE 1.8.11 slow on Pi3 B+

K4ktus123, 2020-01-31:
I just installed the fresh IDE onto my Pi, but I'm pretty dissapointed with its performance - it takes longer to launch and often freezes (when opening library manager for example).
I installed 1.8.10 just to really make sure and it doesn't have issues described above.

per1234, 2020-02-04:
I can reproduce the problem with Library Manager. I have not had it permanently freeze, but it freezes so long, both on opening and closing Library Manager, that you could easily think it was permanent.

I have found that the Library Manager problem goes away after deleting `~/.arduino15/library_index.json` and `~/.arduino15/cache`. Does the same happen for you @K4ktus123?

Once Library Manager opens, do you get the `Error downloading https://downloads.arduino.cc/libraries/library_index.json` error (https://github.com/arduino/Arduino/issues/9719)?

K4ktus123, 2020-02-08:
Yes, I get this error, but when IDE launches.

K4ktus123, 2020-02-08:
Yes, that's exactly what happens.
Also, sorry for taking so long to respond.