arduino/Arduino #9704: External interrupts misfiring

tolletrolden, 2020-02-01:
Looks like attachInterrupt(digitalPinToInterrupt(2), juhu, FALLING); always akt like CHANGE.

No matter what i choose - RISING, FALLING, HIGH or CHANGE it always behave like CHANGE.

Here is the simple sketch:

void setup()

{
pinMode(2, INPUT);
pinMode(3, OUTPUT);

attachInterrupt(digitalPinToInterrupt(2), juhu, RISING);

}

void loop()
{
}

void juhu(){

digitalWrite(3, HIGH);
delayMicroseconds(200);
digitalWrite(3, LOW);
 }

Then top two lines on the scope picture, is a nice clean square wave, made by an optocoupler.
The buttom two lines is the resulting signal on output pin 3.

No matter, if i stand on my head and sing a terrible song, or choose RISING, FALLING or CHANGE, the result is the same - always two signals with one square wave.

Can You tell me what is wrong?

I think the arduino ide translates all modes to CHANGE

[image]

tolletrolden, 2020-02-01:
I tested the triac control from arduino playground - it also akt strange around zero crossing. Also that sketch make 2 dots on the scope, when near zero crossing, but ok when further away from zero crossing.

per1234, 2020-02-01:
Which Arduino board are you using @tolletrolden?

tolletrolden, 2020-02-02:
@per1234 

I have used 2 different uno boards + 1 mega 2560
And finally a stm32f103

All programmed from the arduino ide.

tolletrolden, 2020-02-02:
Now tested ESP32. It seems like it fires the trigger again and again, while input signal is HIGH.
Done some experimenting, but cannot get it right. 

Am I doing something wrong?

[image]

tolletrolden, 2020-02-02:
void setup()
{
  pinMode(2, INPUT);
  pinMode(3, OUTPUT);

  attachInterrupt(digitalPinToInterrupt(2), juhu, RISING);
}

void loop()
{
}

void juhu() {
  delayMicroseconds(1650);
  digitalWrite(3, HIGH);
  delayMicroseconds(200);
  digitalWrite(3, LOW);
}

If i make the above change - ad a delay off 1650us, and get the puls away from zero crossing, then by magic, there is only one pulse.

If the delay is only 1600us then the second pulse flickers on and off.

[image]