arduino/Arduino #9710: "Add .ZIP Library" freezes Microsoft Store version of the Arduino IDE

per1234, 2020-02-02:
Using Arduino IDE 1.8.11 (Windows Store 1.8.28.0) on Windows 10 64 bit (VirtualBox)

1. Start the Windows Store version of the Arduino IDE.
1. **Sketch > Include Library > Add .ZIP Library...** - the dialog does not open and the IDE freezes and must be killed using Task Manager.

The problem only occurs when I launch the IDE from Microsoft Store, or the app pinned to the task bar. It does not occur if I start arduino.exe or arduino_debug.exe directly from `C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt`.

The problem does not occur with the standard Arduino IDE 1.8.11 or 1.8.11 Hourly Build 2020/01/31 09:33

Originally reported at:
- https://forum.arduino.cc/index.php?topic=662171
- https://forum.arduino.cc/index.php?topic=662430

finaldzn, 2020-02-03:
Running into the same problem, not an isolated case

cmaglie, 2020-02-03:
> or the app pinned to the task bar

I'm able to reprodue on the store app but not by pinning. How you pin the app to reproduce the bug?
(I'm asking because it would make the debugging waaay easier)

per1234, 2020-02-03:
@cmaglie:
1. Launch the Arduino IDE from the Microsoft Store.
1. Right click the Arduino IDE icon on the taskbar.
1. Click "Pin to taskbar".
1. Close the Arduino IDE.
1. Click the pinned icon on the taskbar.
1. **Sketch > Include Library > Add .ZIP Library...** - the dialog does not open and the IDE freezes and must be killed using Task Manager.

Let me know if you need any more information.

cmaglie, 2020-02-03:
Ah ok, it's always the MSStore version, I thought it was the downloaded vesion pinned to taskbar.

closed 2020-02-05 by cmaglie

cmaglie, 2020-02-05:
An update with a fix has been sent to the MS Store, it will be available in a couple of hours.