arduino/Arduino #9712: lto1: internal compiler error: in output_constructor_regular_field, at varasm.c:5056

pdmartinson, 2020-02-03:
I get the following error when I think I've resolved all my compile errors. The warning about multiple libraries for "EEPROM.h" appears whether I have compile errors or not.

lto1: internal compiler error: in output_constructor_regular_field, at varasm.c:5056
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
lto-wrapper: fatal error: /home/peter/Documents/Boat/CruiseControl/Code/arduino-1.8.10-linux64/arduino-1.8.10/hardware/tools/avr/bin/avr-gcc returned 1 exit status
compilation terminated.
/home/peter/Documents/Boat/CruiseControl/Code/arduino-1.8.10-linux64/arduino-1.8.10/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
Multiple libraries were found for "EEPROM.h"
 Used: /home/peter/Documents/Boat/CruiseControl/Code/arduino-1.8.10-linux64/arduino-1.8.10/hardware/arduino/avr/libraries/EEPROM
exit status 1
Error compiling for board Arduino/Genuino Uno.

pdmartinson, 2020-02-03:
The version of Arduino is 1.8.10.

facchinm, 2020-02-17:
Hi @pdmartinson ,
can you post a simple sketch that we can use to reproduce the issue? Eg. try removing some functions (even if the sketch doesn't do anything useful afterwards but still fails to compile) and attach it here, thanks.

kiprasmel, 2020-12-27:
Hello, I encountered the same error and have a simple reproduction:

this code gives a different error:

```c

struct A {
	int y[];
	int z[];
};

A a1 = {
	.y = { 1, 2, 3 },
	.z = { 4, 5, 6 }
};

void setup() {

}

void loop() {
	// force usage
	Serial.println(a1.y[0]);
	Serial.println(a1.z[0]);
}
```

```console
Picked up JAVA_TOOL_OPTIONS:
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
boom:3:8: error: flexible array member 'A::y' in an otherwise empty 'struct A'
  int y[];
        ^
exit status 1

[Process exited 1]
```

(see also https://gcc.gnu.org/bugzilla/show_bug.cgi?id=71939)

and with the following diff:

```diff
--- boom.ino
+++ boom.ino
@@ -1,10 +1,12 @@

 struct A {
+       int x;
        int y[];
        int z[];
 };

 A a1 = {
+       .x = 1,
        .y = { 1, 2, 3 },
        .z = { 4, 5, 6 }
 };
@@ -15,6 +17,7 @@

 void loop() {
        // force usage
+       Serial.println(a1.x);
        Serial.println(a1.y[0]);
        Serial.println(a1.z[0]);
 }
```

this code breaks:

```c
struct A {
	int x;
	int y[];
	int z[];
};

A a1 = {
	.x = 1,
	.y = { 1, 2, 3 },
	.z = { 4, 5, 6 }
};

void setup() {

}

void loop() {
	// force usage
	Serial.println(a1.x);
	Serial.println(a1.y[0]);
	Serial.println(a1.z[0]);
}
```

with error

```console
Picked up JAVA_TOOL_OPTIONS:
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
lto1: internal compiler error: in output_constructor_regular_field, at varasm.c:5056
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
lto-wrapper: fatal error: /home/kipras/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc returned 1 exit status
compilation terminated.
/home/kipras/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1

[Process exited 1]
```

---

I'm running arch linux

```sh
$ uname -a
Linux arch-usb2 5.4.85-1-lts #1 SMP Mon, 21 Dec 2020 19:28:35 +0000 x86_64 GNU/Linux
```