arduino/Arduino #9713: Nightly builds are NOT SIGNED - and are available directly form arduino.cc - SECURITY - does anyone care?

Sign-Your-Builds, 2020-02-03:
IoT is well on the way to being the worlds single largest major security issue, propelled by lazy coders who do not care at all for the safety of any of their customers...

IF YOU SHIP EXECUTABLES - THEY SHOULD RUN *WITHOUT* USERS HAVING TO DISABLE SECURITY CHECKS !!!!!!!!

Do not close this issue unless you first either start signing your code, or, stop distributing unsigned code.

PaulStoffregen, 2020-02-03:
Are you using MacOS, or something else?  And which version?  MacOS Catalina requires notarization from Apple in addition to a valid signature.

As far as I know, the Linux and Windows versions should be able to run.

A related (but not duplicate) issue is https://github.com/arduino/arduino-builder/issues/349 which prevents the IDE from passing Apple's notarization process for MacOS.

Sign-Your-Builds, 2020-02-03:
My dev machine is offline, so I'm not sure what version it runs.

Did you *check* to see if any of the hourly/daily releases available on arduino.cc are properly digitally signed?

Regardless of the hoops that Apple forces you to go through to ENSURE YOUR SOFTWARE IS SAFE - those are IMPORTANT and need to be followed properly.  Here's some info.  There's no reason why this can't be integrated into your build - it's very fast:-

https://developer.apple.com/documentation/xcode/notarizing_macos_software_before_distribution

From what I recall - I got the unknown developer warning talking about malware, and it makes me move it to the trash.

I'm part of a group calling for our government to introduce penalties for software developers who disregard security controls.  The UK already has a law on the table about this.  The days where developers could just ignore the safety of their customers is absolutely over - with dire consequences for those who need their "couldn't care less" attitude forcefully adjusted!

I suggest you remove those downloads: make whoever-needs-them fix those security problems if they want to bring them back.

Monday, February 3, 2020, 7:37:57 PM, Paul Stoffregen wrote:

Are you using MacOS, or something else? And which version? MacOS Catalina requires notarization from Apple in addition to a valid signature.
As far as I know, the Linux and Windows versions should be able to run.
A related (but not duplicate) issue is arduino/arduino-builder#349 which prevents the IDE from passing Apple's notarization process for MacOS.

cmaglie, 2020-02-03:
I don't understand your vehemence since the **main official release is compliant**.

As already said elsewhere the Nightly builds are for development purposes only and are not signed, we don't even provide an installer.

> There's no reason why this can't be integrated into your build - it's very fast:-

You're wrong **it's not fast at all** to comply to these guidelines. Just FYI we had to re-build the entire Java JRE internally because the offcial builds of the Java JRE are not done under the Apple SDK 10.9+. Same for the 70% of all the other tools we used and taken for granted for years.

Another gift given by Apple is that the signing tools doesn't work on a head-less console, because you must tick a checkbox on a dialog box at least once, making automation and CI very difficult especially if the CI server is on the cloud.

Saying that we "don't care" is just wrong and out-of-reality considered the amount of efforts we put in this.

PaulStoffregen, 2020-02-03:
To accuse Arduino of "ignore the safety of their customers" is absurd.  They are indeed signing and properly Apple notarizing official releases.

@Sign-Your-Builds - Perhaps **you** are the one here who could use an improvement in attitude?

PaulStoffregen, 2020-02-03:
I can confirm, when I set up signing and notarization here on my Mac, *many* manual steps were required which required GUI-only tools.  After it's all set up, then you can indeed do it all from the command line or a script.  But anyone who believe this is easy to set up (for non-GUI automated builds) has clearly not experienced the process!

tlk, 2020-02-03:
> we had to re-build the entire Java JRE internally because the offcial builds of the Java JRE are not done under the Apple SDK 10.9+.

Hello @cmaglie
Would this also be necessary if the Arduino IDE could be compiled and run with a newer version of Java?

Refs https://github.com/arduino/Arduino/issues/8193

PaulStoffregen, 2020-02-03:
To answer this question, go here and download the latest JRE

https://adoptopenjdk.net/releases.html?variant=openjdk11&jvmVariant=hotspot

Extract it and then run "otool" to check.  Like this:

    otool -l jdk-11.0.6+10-jre/Contents/MacOS/libjli.dylib | grep sdk
    
    otool -l jdk-11.0.6+10-jre/Contents/Home/bin/java | grep sdk

I tried just now and it look like they built it with SDK 10.10.  I didn't carefully check every file, only a quick look at those 2 really important ones.

Also looks like there is absolutely no digital signature.  There's no Contents/_CodeSignature folder (from codesign) or Contents/CodeResources file (from notarization and stapler) inside the bundle.  Also checked just now with this command.

    codesign -vvvv --deep --strict jdk-11.0.6+10-jre/

It prints this:

    jdk-11.0.6+10-jre/: code object is not signed at all
    In architecture: x86_64

I wonder if @Sign-Your-Builds will now go harass the AdoptOpenJDK developers?

tlk, 2020-02-03:
Thanks Paul, that is constructive.

It looks like the AdoptOpenJDK developers are busy working on this! Check out issue 1130 over at https://github.com/AdoptOpenJDK/openjdk-build/issues

(I am not linking directly because of the agressive title and opening of this issue.)

Following your instructions, the nightly builds from https://ci.adoptopenjdk.net/job/build-scripts/job/jobs/job/jdk11u/job/jdk11u-mac-x64-hotspot/ give these results:

```
$ otool -l jdk-11.0.6+9-jre/Contents/MacOS/libjli.dylib | grep sdk
      sdk 10.14
$ otool -l jdk-11.0.6+9-jre/Contents/Home/bin/java | grep sdk
      sdk 10.14
$
```

```
$ codesign -vvvv --deep --strict jdk-11.0.6+9-jre/
jdk-11.0.6+9-jre/: code has no resources but signature indicates they must be present
$ codesign -vvvv --deep --strict jdk-11.0.6+9-jre/Contents/Home/bin/java
jdk-11.0.6+9-jre/Contents/Home/bin/java: valid on disk
$
```

The results for jdk-11.0.6+9 (the JDK) are identical.

cmaglie, 2020-06-24:
Hourly builds are now signed and notarized.

closed 2020-06-24 by cmaglie