arduino/Arduino #9715: COM port not active (handshake?) while uploading sketch via 32U4 arduino

haweiler, 2020-02-04:
Hi, I'm holding on porting JTAG2UPDI for the 32U4. The 32U4 (leonardo) is used for programming ATtiny via UPDI. While receiving data via USB (mapped to COMx) the 32U4 cannot send feedback to the Arduino-IDE (checking comport fails). Maybe a handshake issue on USB->COM? There's no way to disable DTR/DSR, RTS/CTS in the IDE.
When activating the COM-port on a terminal prog (like Realterm) after programming failed all bytes ment as feedback for IDE are sent.
JTAG2UPDI works with 328P devices, so they don't seem to care about handshake and send feedback data to IDE.

haweiler, 2020-02-04:
I've tried to use avrdude now, but with the same result:

C:\Program Files (x86)\Arduino\hardware\tools\avr\etc>..\bin\avrdude -c jtag2updi -P COM3 -p t202
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1

Pressing CTRL-C, then activating port COM3 in Realterm gives you 'sgn_resp' two times.
And while trying to program via jtag2updi the port is locked by avrdude resp Arduino IDE.

haweiler, 2020-02-04:
I took a look if the COM-ports from UNO and Leonardo are installed in a different way.
Both appear as USBSER with USBclass2, subclass 02, Prot 01,
what differs is the device stack:

Leonardo:
\Driver\usbser
\Driver\usbccgp

UNO:
\Driver\usbser
\Driver\USBPcap
\Driver\ACPI
\Driver\USBHUB3

haweiler, 2020-02-05:
To reproduce:
After transfering the sketch, the arduino ends up waiting for the serial to get active:
#elif defined __AVR_ATmega32U4__
	SERIALCOM.begin(19200);   //test 32U4 - baudrate irrelevant
	// wait for Serial to be active
	while (!SERIALCOM){ SYS::LED_blink(5, 1, 100);}; //LED5 flashes 1x for 100ms

Serial is not correctly initialized through Arduino IDE or arvdude. You end up looping endless in the while-loop (LED flashes, and flashes...) until activating port (through opening in realterm), sending the correct control line state request (watch that with USBPcap).

facchinm, 2020-02-17:
For an arduino-compatible jtag2updi you can start from these sources (https://github.com/arduino/ArduinoCore-megaavr/tree/master/firmwares/MuxTO).
I'm closing the issue though since it's not related to Arduino IDE itself but to a SerialUSB behavior (it will probably just work with  the sources I linked).
For any further request, please post on the [forum](https://forum.arduino.cc/) or in [AVR core repo](https://github.com/arduino/ArduinoCore-avr)

closed 2020-02-17 by facchinm