arduino/Arduino #9716: Panic runtime error

jgromes, 2020-02-04:
Hi, I'm running into the following issue when trying to compile any Arduino sketch:

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x834aba]

goroutine 1 [running]:
github.com/arduino/go-properties-orderedmap.(*Map).Get(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/go-properties-orderedmap/properties.go:210
github.com/arduino/arduino-cli/legacy/builder.(*WipeoutBuildPathIfBuildOptionsChanged).Run(0xf47308, 0x1340e780, 0xf47308, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/wipeout_build_path_if_build_options_changed.go:45 +0x15a
github.com/arduino/arduino-cli/legacy/builder.(*ContainerBuildOptions).Run(0xf47308, 0x1340e780, 0xf47308, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_build_options.go:35 +0x115
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x1340e780, 0x13433e1c, 0x20, 0x20, 0x1, 0x2e604fc, 0xfc)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:195 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0x13463f40, 0x1340e780, 0xf47254, 0x13407398)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:102 +0xb9c
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:211
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:399 +0x670
C:\Program Files (x86)\Arduino\arduino-builder returned 2
Error compiling for board Arduino Uno.
```

I'm on Windows 7 (64 bit). It started happening a few hours ago on 1.8.10. Tried updating to 1.8.11 and the latest hourly build (2020/02/03 07:33) but both produce the same error message. Something to note is that my computer shut down due to power outage while compiling - this issue appeared after that.

Any pointers on what to try to fix this would be greatly appreciated.

jgromes, 2020-02-04:
Update - I tried compiling a sketch using standalone 1.8.9 IDE, that seems to be working. After that however, both 1.8.11 and the hourly build compile the sketch correctly.

Not sure whether to close this (as the original issue seems to have been resolved ... somehow) or keep it open for documentation purposes.

cmaglie, 2020-02-04:
It may be a leftover state in the buld folder, it's seems very rare, but it's interesting that 1.8.9 solved it, let's keep it open.