arduino/Arduino #9717: Arduino IDE 1.8.11 not working on Ubuntu 19.10 machine

JulianNicholls, 2020-02-04:
The version 1.8.11 IDE goes haywire on my Ubuntu 19.10 machine. Reverting to IDE version 1.8.10 is fine, though.

It's difficult for me to give much more information because the UI is completely non-functional, i.e. no menus etc. All I get is the two sections, with a huge cursor in the bottom one. I can close it without a problem, but that's all I can do.

cmaglie, 2020-02-04:
Can you provide a screeshot?

heinerolli, 2020-02-06:
I have the exact same problem - on a Windows 10 64 bit machine!
No Text at all - also on the initial splash screen there is no status text ('loading boards..') 
I tried the latest nightly - same problem.....
[image]

cmaglie, 2020-02-06:
Can you run `arduino_debug.exe` from the IDE folder and copy here the output?

heinerolli, 2020-02-06:
`Set log4j store directory C:\arduino\portable
Konfiguration wird geladen...
Pakete werden initialisiert...
DEBUG StatusLogger Using ShutdownCallbackRegistry class org.apache.logging.log4j.core.util.DefaultShutdownCallbackRegistry
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger AsyncLogger.ThreadNameStrategy=UNCACHED (user specified null, default is UNCACHED)
TRACE StatusLogger Using default SystemClock for timestamps.
DEBUG StatusLogger org.apache.logging.log4j.core.util.SystemClock does not support precise timestamps.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Took 0,121049 seconds to load 209 plugins from sun.misc.Launcher$AppClassLoader@af3868
DEBUG StatusLogger PluginManager 'Converter' found 44 plugins
DEBUG StatusLogger Starting OutputStreamManager SYSTEM_OUT.false.false-1
DEBUG StatusLogger Starting LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@16dbea0]...
DEBUG StatusLogger Reconfiguration started for context[name=af3868] at URI null (org.apache.logging.log4j.core.LoggerContext@16dbea0) with optional ClassLoader: null
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger PluginManager 'ConfigurationFactory' found 4 plugins
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Missing dependencies for Yaml support, ConfigurationFactory org.apache.logging.log4j.core.config.yaml.YamlConfigurationFactory is inactive
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Using configurationFactory org.apache.logging.log4j.core.config.ConfigurationFactory$Factory@169afe1
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.properties] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-test.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.properties] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.yml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-test.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
[... cut ...]
DEBUG StatusLogger org.apache.logging.log4j.core.util.SystemClock does not support precise timestamps.
TRACE StatusLogger Using DummyNanoClock for nanosecond timestamps.
DEBUG StatusLogger Reconfiguration complete for context[name=af3868] at URI jar:file:/C:/arduino/lib/pde.jar!/log4j2.xml (org.apache.logging.log4j.core.LoggerContext@16dbea0) with optional ClassLoader: null
DEBUG StatusLogger Shutdown hook enabled. Registering a new one.
DEBUG StatusLogger LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@16dbea0] started OK.
Boards werden vorbereitet...
Startvorgang...`

cmaglie, 2020-02-06:
Unfortunately, everything seem fine in the log.

Did you try the previous release IDE 1.8.10? if yes, is it working as expected?
There is some custom configuration you made on your system that you think may affect the interface (like Fonts, Colors, Themes, Languages)?

heinerolli, 2020-02-06:
Just copied 1.8.10 over again and that works w/o issues. I don*t think I did anything with the configuration. It might be that I tried something with a dark theme, but I reverted that back a while ago

cmaglie, 2020-02-06:
May you make the following test:

- delete the `java` folder from the IDE 1.8.11
- copy the `java` folder from IDE 1.8.10 to IDE 1.8.11
- run IDE 1.8.11

I fear that it may a problem related to the latest Java.
Thanks for your time!

heinerolli, 2020-02-06:
Yes, you are right! 
Doing the above 1.8.11 runs w/o issues