arduino/Arduino #9719: "signature check failed" when downloading Library and Boards Manager indexes on Raspberry Pi

per1234, 2020-02-04:
Using Arduino IDE 1.8.11 Hourly Build 2020/02/03 07:20 on Raspbian 9.4 (stretch) and Raspbian 10.2 (buster)

1. Delete `~/.arduino15/cache` (because if the Arduino IDE uses the cached indexes, it won't trigger the bug)
1. (OPTIONAL) If you want to avoid waiting over 5 minutes for Library Manager to open (due to https://github.com/arduino/Arduino/issues/9701), delete `~/.arduino15/library_index.json`. The bug occurs with or without it, it just occurs a lot faster without.
1. Start the Arduino IDE
1. **Sketch > Include Library > Manage Libraries...**

The index download fails:
```
 Error downloading https://downloads.arduino.cc/libraries/library_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading https://downloads.arduino.cc/libraries/library_index.json
    at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onUpdatePressed$2(LibraryManagerUI.java:207)
    at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://downloads.arduino.cc/libraries/library_index.json
    at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
    at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:130)
    at cc.arduino.contributions.JsonDownloader.download(JsonDownloader.java:49)
    at cc.arduino.contributions.GZippedJsonDownloader.download(GZippedJsonDownloader.java:66)
    at cc.arduino.contributions.libraries.LibraryInstaller.updateIndex(LibraryInstaller.java:84)
    at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onUpdatePressed$2(LibraryManagerUI.java:203)
    ... 1 more
Caused by: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: signature check failed
    at sun.security.ssl.Alerts.getSSLException(Alerts.java:192)
    at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1946)
    at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:316)
    at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:310)
    at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1639)
    at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:223)
    at sun.security.ssl.Handshaker.processLoop(Handshaker.java:1037)
    at sun.security.ssl.Handshaker.process_record(Handshaker.java:965)
    at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1064)
    at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1367)
    at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1395)
    at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1379)
    at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559)
    at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
    at sun.net.www.protocol.https.HttpsURLConnectionImpl.connect(HttpsURLConnectionImpl.java:167)
    at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:150)
    at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:106)
    at cc.arduino.utils.network.FileDownloaderCache.updateCacheInfo(FileDownloaderCache.java:184)
    at cc.arduino.utils.network.FileDownloaderCache.getFileCached(FileDownloaderCache.java:153)
    at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:167)
    at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
    at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
    ... 6 more
Caused by: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: signature check failed
    at sun.security.validator.PKIXValidator.doValidate(PKIXValidator.java:362)
    at sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:259)
    at sun.security.validator.Validator.validate(Validator.java:262)
    at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:330)
    at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:237)
    at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:132)
    at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1621)
    ... 23 more
Caused by: java.security.cert.CertPathValidatorException: signature check failed
    at sun.security.provider.certpath.PKIXMasterCertPathValidator.validate(PKIXMasterCertPathValidator.java:135)
    at sun.security.provider.certpath.PKIXCertPathValidator.validate(PKIXCertPathValidator.java:233)
    at sun.security.provider.certpath.PKIXCertPathValidator.validate(PKIXCertPathValidator.java:141)
    at sun.security.provider.certpath.PKIXCertPathValidator.engineValidate(PKIXCertPathValidator.java:80)
    at java.security.cert.CertPathValidator.validate(CertPathValidator.java:292)
    at sun.security.validator.PKIXValidator.doValidate(PKIXValidator.java:357)
    ... 29 more
Caused by: java.security.SignatureException: Signature does not match.
    at sun.security.x509.X509CertImpl.verify(X509CertImpl.java:424)
    at sun.security.provider.certpath.BasicChecker.verifySignature(BasicChecker.java:166)
    at sun.security.provider.certpath.BasicChecker.check(BasicChecker.java:147)
    at sun.security.provider.certpath.PKIXMasterCertPathValidator.validate(PKIXMasterCertPathValidator.java:125)
    ... 34 more
```

[... 476 lines cut ...]
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:106) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloaderCache.updateCacheInfo(FileDownloaderCache.java:184) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloaderCache.getFileCached(FileDownloaderCache.java:153) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:167) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129) ~[arduino-core.jar:?]
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147) ~[arduino-core.jar:?]
	... 4 more
2020-02-04T15:28:33.339Z INFO c.a.c.p.ContributionInstaller:314 [ContributionManager Update Thread] Downloaded package index URL=[https://downloads.arduino.cc/packages/package_index.json]
2020-02-04T15:28:33.345Z INFO c.a.c.p.ContributionInstaller:324 [ContributionManager Update Thread] Check unknown files. Additional package index folder files=[package_index.json], Additional package index url downloaded=[]
```
</details>

---
The issue does not occur with Arduino IDE 1.8.10.

The issue does occur with Arduino IDE 1.8.11.

The issue does not occur with Windows 10 64 bit or Linux 64 bit versions of the Arduino IDE.

---
Originally reported at:
- https://forum.arduino.cc/index.php?topic=661984
- https://forum.arduino.cc/index.php?topic=662241
- https://forum.arduino.cc/index.php?topic=661342.msg4464663#msg4464663
- And I believe also: https://github.com/arduino/Arduino/issues/8474#issuecomment-580869254, where @twling reported the results of their investigation

tlk, 2020-02-04:
Testing with SSLPoke from https://gist.github.com/4ndrej/4547029

OpenJDK from Raspbian Buster:
```
$ uname -a
Linux pi.local 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux
$ lsb_release -d
Description:	Raspbian GNU/Linux 10 (buster)
$ sudo apt install openjdk-11-jdk-headless openjdk-11-jre-headless ca-certificates-java

$ /usr/bin/java -version
openjdk version "11.0.6" 2020-01-14
OpenJDK Runtime Environment (build 11.0.6+10-post-Raspbian-1deb10u1)
OpenJDK Server VM (build 11.0.6+10-post-Raspbian-1deb10u1, mixed mode)
$ /usr/bin/javac -version
javac 11.0.6
$
$ sudo update-ca-certificates
$ ls -la /etc/ssl/certs/java/cacerts
-rw-r--r-- 1 root root 151481 Feb  4 22:46 /etc/ssl/certs/java/cacerts
$ sha256sum /etc/ssl/certs/java/cacerts
12128caf5a772e67af5986cea0f808e4a13f70101e6865e3f82e5d9f66998b48  /etc/ssl/certs/java/cacerts
$ wget -q https://gist.githubusercontent.com/4ndrej/4547029/raw/84d3bff7bba262b3f77baa32a43873ea95993e45/SSLPoke.java
$ /usr/bin/javac --release 8 SSLPoke.java SSLPoke.java
$ /usr/bin/java -Djavax.net.ssl.trustStore=/etc/ssl/certs/java/cacerts SSLPoke downloads.arduino.cc 443
Successfully connected
$
```

Good.

Now let us test the JRE provided in arduino-1.8.11-linuxarm.tar.xz:

```
$ wget -q https://downloads.arduino.cc/arduino-1.8.11-linuxarm.tar.xz
$ du -sh arduino-1.8.11-linuxarm.tar.xz
88M	arduino-1.8.11-linuxarm.tar.xz
$ sha256sum arduino-1.8.11-linuxarm.tar.xz
6c82471c47b3c3a5efd6610115f0fcb7f7eec69d19e1ee5e7882db72634a1611  arduino-1.8.11-linuxarm.tar.xz
$ tar xJf arduino-1.8.11-linuxarm.tar.xz
$ ./arduino-1.8.11/java/bin/java SSLPoke downloads.arduino.cc 443
javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: signature check failed
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:192)
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1946)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:316)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:310)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1639)
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:223)
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:1037)
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:965)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1064)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1367)
	at sun.security.ssl.SSLSocketImpl.writeRecord(SSLSocketImpl.java:750)
	at sun.security.ssl.AppOutputStream.write(AppOutputStream.java:123)
	at sun.security.ssl.AppOutputStream.write(AppOutputStream.java:138)
	at SSLPoke.main(SSLPoke.java:28)
Caused by: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: signature check failed
	at sun.security.validator.PKIXValidator.doValidate(PKIXValidator.java:362)
	at sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:259)
	at sun.security.validator.Validator.validate(Validator.java:262)
	at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:330)
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:237)
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:132)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1621)
	... 9 more
Caused by: java.security.cert.CertPathValidatorException: signature check failed
	at sun.security.provider.certpath.PKIXMasterCertPathValidator.validate(PKIXMasterCertPathValidator.java:135)
	at sun.security.provider.certpath.PKIXCertPathValidator.validate(PKIXCertPathValidator.java:233)
	at sun.security.provider.certpath.PKIXCertPathValidator.validate(PKIXCertPathValidator.java:141)
	at sun.security.provider.certpath.PKIXCertPathValidator.engineValidate(PKIXCertPathValidator.java:80)
	at java.security.cert.CertPathValidator.validate(CertPathValidator.java:292)
	at sun.security.validator.PKIXValidator.doValidate(PKIXValidator.java:357)
	... 15 more
Caused by: java.security.SignatureException: Signature does not match.
	at sun.security.x509.X509CertImpl.verify(X509CertImpl.java:449)
	at sun.security.provider.certpath.BasicChecker.verifySignature(BasicChecker.java:166)
	at sun.security.provider.certpath.BasicChecker.check(BasicChecker.java:147)
	at sun.security.provider.certpath.PKIXMasterCertPathValidator.validate(PKIXMasterCertPathValidator.java:125)
	... 20 more
$
```

Same results when using the cacert from Raspbian Buster (ca-certificates-java):
```
$ ./arduino-1.8.11/java/bin/java -Djavax.net.ssl.trustStore=/etc/ssl/certs/java/cacerts SSLPoke downloads.arduino.cc 443
```

tlk, 2020-02-04:
FWIW, same error when testing against cloudflare.com:

```
$ ./arduino-1.8.11/java/bin/java SSLPoke cloudflare.com 443
javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: signature check failed
   <..>
$
```

cmaglie, 2020-02-04:
@tlk  thanks for your tests.

Moving to JRE 11 will create another set of problems, but it's something that we planned to do in the near future.

May you try the Java 8 version (1.8.0_232)  here https://adoptopenjdk.net/releases.html?

tlk, 2020-02-05:
You are welcome!

Same results with Java 8 version (1.8.0_232) from https://adoptopenjdk.net/releases.html:
```
$ ./jdk8u232-b09-jre/bin/java SSLPoke downloads.arduino.cc 443
javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: signature check failed
   <..>
$
```

tlk, 2020-02-05:
Loading https://downloads.arduino.cc in Chrome version 79.0.3945.130 it reports a certificate chain where the root certificate (common name "COMODO ECC Certification Authority") has this SHA-256 fingerprint:

```
17 93 92 7A 06 14 54 97 89 AD CE 2F 8F 34 F7 F0 B6 6D 0F 3A E3 A3 B8 4D 21 EC 15 DB BA 4F AD C7
```

Checking the trustStores both have a match, so that looks good:
```
$ keytool -list -cacerts -storepass changeit | grep 17:93:92:7A:06:14:54:97:89:AD:CE:2F:8F:34:F7:F0:B6:6D:0F:3A:E3:A3:B8:4D:21:EC:15:DB:BA:4F:AD:C7
Certificate fingerprint (SHA-256): 17:93:92:7A:06:14:54:97:89:AD:CE:2F:8F:34:F7:F0:B6:6D:0F:3A:E3:A3:B8:4D:21:EC:15:DB:BA:4F:AD:C7
$ keytool -list -storepass changeit -keystore ./arduino-1.8.11/java/lib/security/cacerts | grep 17:93:92:7A:06:14:54:97:89:AD:CE:2F:8F:34:F7:F0:B6:6D:0F:3A:E3:A3:B8:4D:21:EC:15:DB:BA:4F:AD:C7
Certificate fingerprint (SHA-256): 17:93:92:7A:06:14:54:97:89:AD:CE:2F:8F:34:F7:F0:B6:6D:0F:3A:E3:A3:B8:4D:21:EC:15:DB:BA:4F:AD:C7
$
```

However, debugging with openssl it reports a certificate chain where the root certificate (same common name "COMODO ECC Certification Authority") has a different SHA-256 fingerprint:
```
95:73:86:2A:C0:B4:B1:25:16:88:10:EA:3F:D1:01:AE:2E:B0:BB:15:F6:1F:C0:E6:DA:7A:2A:38:B8:5A:89:E8
```

There is probably a better way to fetch the fingerprints with openssl, but I took https://stackoverflow.com/a/44207749/936466 and adjusted it so it will fetch the certificate chain and print common name and SHA-256 fingerprint in pairs:
```
$ openssl s_client -showcerts -connect downloads.arduino.cc:443  < /dev/null 2>/dev/null \
    | sed -ne '/-BEGIN CERTIFICATE-/,/-END CERTIFICATE-/p;/-END CERTIFICATE-/a\\x0' \
    | sed -e '$ d' \
    | xargs -0rl -I% \
        sh -c "echo '%' | openssl x509 -in /dev/stdin -sha256 -text -fingerprint" \
    | grep -e Subject: -e ^SHA256 \
    | sed -e 's/.* CN = /CN = /'

CN = ssl788311.cloudflaressl.com
SHA256 Fingerprint=47:5F:1D:B8:BC:77:F1:E1:96:30:F6:B5:EB:0A:22:DD:F7:51:2E:3D:3E:F3:A1:DC:E9:30:B4:19:68:9B:C7:57
CN = COMODO ECC Domain Validation Secure Server CA 2
SHA256 Fingerprint=CD:6C:10:8A:0E:64:1F:2C:A1:22:AA:A6:D0:3F:82:67:59:CA:E7:C6:F8:00:EA:BF:76:DC:48:B6:7C:D0:83:CE
CN = COMODO ECC Certification Authority
SHA256 Fingerprint=95:73:86:2A:C0:B4:B1:25:16:88:10:EA:3F:D1:01:AE:2E:B0:BB:15:F6:1F:C0:E6:DA:7A:2A:38:B8:5A:89:E8
$
```

Looking closer (at the output from the same command but with the last grep and sed removed) shows that the CN = ssl788311.cloudflaressl.com certificate has this info:
```
            X509v3 Subject Alternative Name:
                DNS:ssl788311.cloudflaressl.com, DNS:*.arduino.cc, DNS:arduino.cc
```

Ok.

Checking the fingerprints of the first two certificates (47:5F.. and CD:6C..) Chrome reports identical fingerprints.

But while Chrome reports a 17:93.. fingerprint for the root cert openssl reports a 95:73.. fingerprint. Odd.

Well, let's check the trustStores for this 95:73.. fingerprint:

$ keytool -list -cacerts -storepass changeit | grep 95:73:86:2A:C0:B4:B1:25:16:88:10:EA:3F:D1:01:AE:2E:B0:BB:15:F6:1F:C0:E6:DA:7A:2A:38:
$ keytool -list -storepass changeit -keystore ./arduino-1.8.11/java/lib/security/cacerts | grep 95:73:86:2A:C0:B4:B1:25:16:88:10:EA:3F:D1:01:AE:2E:B0:BB:15:F6:1F:C0:E6:DA:7A:2A:38:
$

No match!

I must have missed something because why would openssl and OpenJDK from Raspbian Buster happily accept a certificate chain with an unknown root cert?

Let's have a closer look at that root cert with the 95:73.. fingerprint:

```
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            43:52:02:3f:fa:a8:90:1f:13:9f:e3:f4:e5:c1:44:4e
        Signature Algorithm: sha384WithRSAEncryption
        Issuer: C = SE, O = AddTrust AB, OU = AddTrust External TTP Network, CN = AddTrust External CA Root
        Validity
            Not Before: May 30 10:48:38 2000 GMT
            Not After : May 30 10:48:38 2020 GMT
        Subject: C = GB, ST = Greater Manchester, L = Salford, O = COMODO CA Limited, CN = COMODO ECC Certification Authority
```

Okay, so the "Issuer:" line claims that it was issued by a certificate with CN = AddTrust External CA Root.

Maybe we can find that in the trustStores?

```
$ keytool -cacerts -storepass changeit -exportcert -alias debian:addtrust_external_root.pem -rfc | openssl x509 -in /dev/stdin -sha256 -text -fingerprint | grep -e Subject: -e ^SHA256
        Subject: C = SE, O = AddTrust AB, OU = AddTrust External TTP Network, CN = AddTrust External CA Root
SHA256 Fingerprint=68:7F:A4:51:38:22:78:FF:F0:C8:B1:1F:8D:43:D5:76:67:1C:6E:B2:BC:EA:B4:13:FB:83:D9:65:D0:6D:2F:F2
$
$ keytool -storepass changeit -keystore ./arduino-1.8.11/java/lib/security/cacerts -exportcert -alias "addtrustexternalca [jdk]" -rfc | openssl x509 -in /dev/stdin -sha256 -text -fingerprint | grep -e Subject: -e ^SHA256
        Subject: C = SE, O = AddTrust AB, OU = AddTrust External TTP Network, CN = AddTrust External CA Root
SHA256 Fingerprint=68:7F:A4:51:38:22:78:FF:F0:C8:B1:1F:8D:43:D5:76:67:1C:6E:B2:BC:EA:B4:13:FB:83:D9:65:D0:6D:2F:F2
$
```

Yup! matching fingerprints!

I wonder if the 95:73.. certificate is properly signed by the "AddTrust External CA Root" certificate, and if so - why this is not accepted by the Java 8 version (1.8.0_232) from AdoptOpenJDK.

To be continued...

tlk, 2020-02-05:
Ok, the fingerprints are not used in the chain validation (AFAIK) and the DN/CN, Serial number and X509v3 key identifiers are identical on the two certificates, so the two different fingerprints should be ok.

I don't understand why the various Java versions from AdoptOpenJDK doesn't work, but on the bright side the issue is not limited to the downloads.arduino.cc domain and it is relatively easy to reproduce with SSLPoke.

At first I thought the errors were caused by the AdoptOpenJDK cacert-files containing fewer CA root certificates than the Raspbian cacert. If that was the case the issue should be solved by using another cacert-file, but that does not help.

cmaglie, 2020-02-06:
I've tested the following JREs using your SSLPoke example:

JRE  | arch | exact version | Builder  |  Working
------ | --- | -------- | --------- | ------------
1.8.0_232 (in IDE) | arm32 | 1.8.0_232-b09 | AdoptOpenJDK | NO
[1.8.0_232](https://github.com/AdoptOpenJDK/openjdk8-binaries/releases/download/jdk8u232-b09/OpenJDK8U-jre_x64_linux_hotspot_8u232b09.tar.gz) | x64 | 1.8.0_232-b09 | AdoptOpenJDK | YES (!)
1.8.0_212 (apt-get) | arm32 | 1.8.0_212-8u212-b01-1+rpi1-b01 | Raspbian | YES
[1.8.0_222](https://github.com/AdoptOpenJDK/openjdk8-binaries/releases/download/jdk8u222-b10/OpenJDK8U-jre_arm_linux_hotspot_8u222b10.tar.gz) | arm32 | 1.8.0_222-b10 | AdoptOpenJDK | NO
[1.8.0_252](https://github.com/AdoptOpenJDK/openjdk8-binaries/releases/download/jdk8u-2020-02-06-13-39/OpenJDK8U-jre_arm_linux_hotspot_2020-02-06-13-39.tar.gz) | arm32 | 1.8.0_252-202002060440-b01 | AdoptOpenJDK | NO

There are no AdoptOpenJDK builds for arm32 before 1.8.0_222 and the latest available from raspbian is 1.8.0_212.
Looking at the table above, at a first sight, I thought that a regression happened between 1.8.0_212 and 1.8.0_222 BUT the same release 1.8.0_232 for x64 is WORKING, so the only reasonable conclusion is that the builds made from AdoptOpenJDK for ARM 32 are buggy.

Is Raspbian applying patches? Is the java version made by raspian redistributable? In that case we may consider to bundle that one.