arduino/Arduino #9720: Add IRSense library to Library Manager

ndholmes, 2020-02-04:
Here's the repository:
https://github.com/IowaScaledEngineering/arduino-irsense/

It's my first submission to the library manager, and I think I've got everything in correct order for submitting this, but if I've screwed this up somehow please let me know and I'll correct it.

per1234, 2020-02-05:
Hi @ndholmes. There are a couple problems with the `name` value in your library.properties that will block adding the library to the Arduino Library Manager index:

Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries. Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces ( ), underscores (_), dots (.) and dashes (-). It cannot start or end with a space, and also it cannot start with a number. Your library name contains the disallowed character `/`.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

ndholmes, 2020-02-06:
Sorry about that, and thanks for the clear guidance.  I had missed those details in the name element.  I think I've got everything fixed up now to meet standards with version 1.0.1.

cmaglie, 2020-02-07:
Done.

closed 2020-02-07 by cmaglie