arduino/Arduino #9723: Every time I go to the editor, it says I have to reinstall the app (preventing my from uploading to my arduino board), even though I have already installed it.

ZaiderBaiter, 2020-02-05:
In my school, there is a class called STEM 2. In STEM 2, we learn to code and wire projects using arduino create and arduino circuits. But every time our class tries to use arduino, at least one person has this issue. My teacher will have to totally remove us from her class in arduino, and re-add us back. While she is doing that, we have to uninstall/reinstall the app. This has never happened to me or my teacher before the last few weeks, so it seems this bug is fairly new. My school email and my teachers email are below.

18948@usd253.net  (my email)

Amanda.Schild@usd253.net  (my teachers email)

[image]

<iframe src="https://drive.google.com/file/d/12hymBSBrE8nfB91wnI-bRo_6cdZp1324/preview" width="640" height="480"></iframe>

ZaiderBaiter, 2020-02-05:
Btw add the part at the bottom in an html editor to see the video.

per1234, 2020-02-05:
Hi @ZaiderBaiter. I'm sorry to hear you're having a problem with the Arduino Create Chrome app.

I'm closing this issue because the correct place to report issues related to the Arduino Create Chrome app is the Arduino forum's **Create > Chrome App** board:
https://forum.arduino.cc/index.php?board=112.0
If you haven't done so already, please make a post there. Thanks!

closed 2020-02-05 by per1234