arduino/Arduino #9724: Bundled Arduino AVR Boards used after installing different version via Boards Manager

per1234, 2020-02-05:
Using Arduino IDE 1.8.11 Hourly Build 2020/02/05 06:33 with Windows 10 64 bit and Ubuntu 19.10 64 bit

1. Use Boards Manager to install a previous version of Arduino AVR Boards.
1. File > Preferences > Show verbose output during: compilation (check) > OK
1. Sketch > Verify/Compile

After the compilation finishes, examine the output in the console window. You can see that the Arduino AVR Boards core bundled with the Arduino IDE is still being used, when it should be using the one installed to the `.arduino15` folder.

Here's a snippet of the output with Arduino AVR Boards 1.6.21 installed:
```
/home/per/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10811 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/home/per/ArduinoIDE/arduino-nightly/hardware/arduino/avr/cores/arduino -I/home/per/ArduinoIDE/arduino-nightly/hardware/arduino/avr/variants/mega /tmp/arduino_build_348441/sketch/sketch_feb05a.ino.cpp -o /dev/null
```
Note that, correctly, avr-gcc 4.9.2-atmel3.5.4-arduino2 is being used, as specified for Arduino AVR Boards 1.6.21 in package_index.json.

However, the `-I` option for the core library folder is:
```
-I/home/per/ArduinoIDE/arduino-nightly/hardware/arduino/avr/cores/arduino
```
when it should be:
```
-I/home/per/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino
```
The issue is not restricted to the `-I` option for the core library folder. That is just a clearly visible indicator of the problem.

---
The issue does occur with Arduino IDE 1.8.11.

The issue does not occur with Arduino IDE 1.8.10

facchinm, 2020-02-11:
It is due to a regression in builder (see https://github.com/arduino/arduino-cli/blob/master/arduino/cores/packagemanager/package_manager.go#L144 and https://github.com/arduino/arduino-cli/blob/d83170eb106daf71079174deb857063bc6cc1dd7/legacy/builder/target_board_resolver.go#L31).
`pm.Packages` gets populated with `avr` key (shared between bundled and board manager one) but tools are passed by the java IDE via rewrite keys.
Can we rethink about how  `pm.Packages` is populated? @masci @cmaglie

facchinm, 2020-02-11:
Inverting https://github.com/arduino/arduino-cli/blob/d83170eb106daf71079174deb857063bc6cc1dd7/arduino/cores/packagemanager/loader.go#L168 condition should solve the issue (board manager cores would get the priority, and tools substitution would work)

cmaglie, 2020-02-11:
Inverting the condition woudn't change the outcome, but your analysis is right we must change the order of calls on `loadPlatforms`.

closed 2020-02-12 by facchinm