arduino/Arduino #9733: No longer possible to install libraries from a folder using "Add .ZIP Library"

per1234, 2020-02-07:
Using Arduino IDE 1.8.11 Hourly Build 2020/02/07 07:33 with Windows 10 64 bits and Ubuntu 19.10

1. Sketch > Include Library > Add .ZIP Library...
1. Select a folder containing a library you want to install.
1. Click the "Open" button.

Expected behavior:
The library is installed.

Actual behavior:
The selected folder is opened in the "Add .ZIP Library..." dialog.

---
Perhaps the ability to install libraries from folders via "Add .ZIP Library..." is not an official usage of that feature, but it is very useful. 

In order to install a library from a .zip file, the library must be in the root of the .zip file, not in a subfolder. But sometimes library .zip files are not structured this way (e.g., https://github.com/RobTillaart/Arduino, https://github.com/jrowberg/i2cdevlib).

The alternative is a manual installation, but beginners often have a difficult time correctly installing libraries manually, and many have almost no experience working directly with the file system.

"Add .ZIP Library" is great because it's easier to use, and ensures the library is correctly installed.

---
The issue also occurs with Arduino IDE 1.8.11 Hourly Build 2020/02/05 06:33.

The issue does not occur with Arduino IDE 1.8.11.

---
I suspect the changed behavior was caused by https://github.com/arduino/Arduino/commit/7cb14400068529bbb917241880a92ab918a4d7bd

cmaglie, 2020-02-07:
This is very bad, I've checked and there is no way to let `FileDialog` select a directory *and* a file.
We can revert 7cb1440 but then the Windows Store version will fail...

I didn't thought that OpenJDK (and in particular the build from AdoptOpenJDK) would be so problematic.

cmaglie, 2020-02-11:
Reverted in 678420c

closed 2020-02-11 by cmaglie