arduino/Arduino #9736: fork/exec {runtime.tools.python.path}/python: no such file or directory

khushvir, 2020-02-08:
fork/exec {runtime.tools.python.path}/python: no such file or directory
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

How to resolve this issue.

matthijskooijman, 2020-02-08:
How did you install the NodeMCU core? Through the boards manager? What Arduino version are you using? What NodeMCU core version?

khushvir, 2020-02-08:
I have installed it through the boards manager.
Arduino IDE version 1.8.11
core 2.6.3

I was following below steps mentioned here --->https://github.com/Azure/azure-iot-arduino

I think some issue with "platform.local.txt" file. But I don't know what's the exact issue here.

matthijskooijman, 2020-02-08:
Ah, it seems you made quite some changes to the default ESP core that likely cause this. Could you try reverting all those (probably just reinstalling the core)? If compiling a simple empty sketch works with a clean ESP core, but not with the additions from that page, then you should probably file an issue with that library (we cannot provide support for all third-party stuff, especially such invasive changes that are hard for us to oversee).

per1234, 2020-02-08:
As @matthijskooijman said, this is off topic for this repository, so I'm closing this. We'll be happy to help you fix the problem if you post about it on the Arduino Forum though:
https://forum.arduino.cc/

closed 2020-02-08 by per1234