arduino/Arduino #9737: SoftwareSerial isn't working in Arduino Beetle

isfanazha, 2020-02-08:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

matthijskooijman, 2020-02-08:
Can you provide some more info? What is the problem exactly? Any error messages? What IDE version? What board do you have selected in the IDE?

What board are you using exactly? Do you mean the DFRobot Beetle?

isfanazha, 2020-02-08:
Yes @matthijskooijman, it's DFRobot Beetle. There is no error in IDE, my version is Arduino 1.8.10. In IDE, I choose Arduino Leonardo board.

per1234, 2020-03-05:
@isfanazha this is still not enough information for us to be able to investigate your issue.

Please provide:
- Minimal, complete sketch that demonstrates the issue.
- Your circuit.
- Detailed description of the behavior you were expecting and the behavior that you observed that did not match your expectations.

isfanazha, 2020-03-07:
@per1234, I just follow tutorial from https://www.arduino.cc/en/Tutorial/SoftwareSerialExample and change the RX and TX pin based on [DFRobot Beetle wiki](https://wiki.dfrobot.com/Beetle_SKU_DFR0282). When I send messages using serial monitor, it makes Arduino IDE freeze, I need to disconnect the DFRobot Beetle to solve that freeze.

- Code
```
#include <SoftwareSerial.h>

SoftwareSerial mySerial(0, 1); // RX, TX

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Native USB only
  }

  Serial.println("Goodnight moon!");

  // set the data rate for the SoftwareSerial port
  mySerial.begin(38400);
  mySerial.println("Hello, world?");
}

void loop() // run over and over
{
  if (mySerial.available())
    Serial.write(mySerial.read());
  if (Serial.available())
    mySerial.write(Serial.read());
}
```

per1234, 2021-04-02:
This seems very much like https://github.com/arduino/Arduino/issues/11122 to me, but this sketch does clear the RX buffer and I'm not able to reproduce https://github.com/arduino/Arduino/issues/11122 with this sketch running on my Leonardo with the stock 1.8.10 installation (bundled Arduino AVR Boards 1.8.1).

@isfanazha please try again with the latest hourly build of the Arduino IDE to see if it has already been fixed and let us know what the results are:
https://www.arduino.cc/en/software#hourly-builds

closed 2022-06-01 by per1234