arduino/Arduino #9738: Arduino 1.8.11 Library Manager: Comparison method violates its original contract.

rei-vilo, 2020-02-08:
On the Arduino 1.8.11 IDE, when calling the menu **Sketch > Include Library > Manage Libraries** and asking the list of updatable librairies, the **Library Manager** displays the following error message:

> Comparison method violates its original contract.

[image]

+ What does it mean?
+ How to get rid of it?

Thank you!

cmaglie, 2020-02-12:
A possible explanation is here: #9748 

May you try the pull-requrest build to see if it solves the problem?

rei-vilo, 2020-02-12:
Thank you! `arduino-PR-9748-BUILD-918-macosx` seems to solve the issue with the Library Manager.

closed 2020-02-12 by cmaglie

BigGene, 2021-11-09:
Just had this error using Arduino IDE 1.8.13.

MortenHofft, 2022-02-08:
I just installed the software (Arduino IDE 1.8.19 downloaded from https://www.arduino.cc/en/software) for the first time and sees this as the first thing.

[image]

OS: Catalina 10.15.7

PR #9748 (2020-02-12): Fix libraries comparator logic in library manager
Previously it could lead to "contract violations", for example consider this:

A = Library{ Name:"A", Types: ["Sensors"] }
B = Library{ Name:"B", Types: null        }
C = Library{ Name:"C", Types: ["Arduino"] }

it results in:

A<B (because B has Types==null and compareName("A","B")<0)
B<C (because B has Types==null and compareName("B","C")<0)
C<A (becuase C has Types=="Arduino" and the comparator returns -1 directly)

This commit fix this behavior.

Fix #9738