arduino/Arduino #9740: [Documentation] I2C contains "duplicate" code

branisha, 2020-02-09:
Following website: https://www.arduino.cc/en/Tutorial/MasterReader contains code both as master receiver and as slave receiver.

Correct would be master receiver and slave sender/transmitter(File->Examples->Wire->slave_sender).
[image]

per1234, 2020-02-09:
The Wire library's slave_sender example sketch:
https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/Wire/examples/slave_sender/slave_sender.ino
is the code that should be in the tutorial in place of the slave_receiver sketch. However, one change needs to be made. The tutorial is written to use I2C address `2`:
```c++
Wire.requestFrom(2, 6);    // request 6 bytes from slave device #2
```
but the Wire library's examples use I2C address `8`:
```c++
Wire.begin(8);                // join i2c bus with address #8
```
My recommendation is, rather than changing the slave_sender sketch to use address `2`, to change the tutorial's master_reader sketch to use address `8`. This will make porting any future changes that might be made in the Wire library's examples to the tutorial easier and less error-prone.

deepaklorkhatri007, 2020-02-29:
Hi @per1234, this issue seems not that complicated to resolve. I would like to help but I am unable to find any direct repo OR any other way to change the content for Tutorials on Examples from Libraries. If there is any way I can help please let me know :)

per1234, 2020-02-29:
@deepaklorkhatri007 unfortunately, the change needs to be made in a place that is not publicly available, so it's not possible for you to help with this. Thanks for your interest in contributing though. If you want to help improve Arduino's documentation, the entire [Arduino Language Reference](https://www.arduino.cc/reference/en/) is available for pull requests and issue reports here:
https://github.com/arduino/reference-en
We also have repositories for translating the language reference into several languages other than English:
https://github.com/arduino?q=reference

deepaklorkhatri007, 2020-02-29:
Thanks for the information @per1234, I will definitely look into the language reference issues :)

ijash, 2020-03-10:
@per1234  hi, any hints when this gonna be fixed?

per1234, 2020-05-24:
Hi @ijash. Sorry for the delay in responding.
>@per1234 hi, any hints when this gonna be fixed?

Unfortunately, I don't know the answer to this. I have no involvement with maintenance of, or access to, this content.

>also i want to inform about this page (should this be separate issue?):

Thanks for bringing this to our attention! Please open a dedicated issue for the missing code and broken links on https://www.arduino.cc/en/Tutorial/SlaveSender.

per1234, 2020-09-28:
>Correct would be master receiver and slave sender/transmitter(File->Examples->Wire->slave_sender).

This has now been fixed.

>My recommendation is, rather than changing the slave_sender sketch to use address 2, to change the tutorial's master_reader sketch to use address 8.

The tutorial now makes no explicit mention of an I2C address, so there is not conflict with the address 8 used in the example sketches of the tutorial.

>also i want to inform about this page (should this be separate issue?):

The "Master Reader/Slave Sender" tutorial now covers both example sketches, so this page no longer exists.

---
Since it looks like everything discussed here has been resolved, I'll go ahead and close it.

closed 2020-09-28 by per1234