arduino/Arduino #9742: Arduino 1.8.11-32bits fails to launch on Lubuntu 18.04: could not open jvm.cfg

AtelierMadman, 2020-02-10:
Installing arduino-1.8.11 32bit (dnloaded form arduino.cc) on my Lubuntu 18.04 laptop.

The install process runs fine, but lauching arduino (in terminal or by clicking on icon) fails.

- ./arduino
    - terminal says "Error : could not open /opt/arduino-1.8.11/java/lib/i386/jvm.cfg"

- Searched for such a file,
     - found one in /usr/lib/jvm/java-1.11.0-openjdk-i386/lib
     - copied it over go /opt/arduino-1.8.11/java/lib

- ./arduino
     - terminal says "Error : failed /opt/arduino-1.8.11/java/lib/i386/server/libjvm.so, because /lib/i386-linux-gnu/libm.so.6: version 'GLIBC 2.29' not found (required by /opt/arduino-1.8.11/java/lib/i386/server/libjvm.so)"

hcremer, 2020-02-10:
Same problem with arduino-1.8.11 Linux 32bit on Debian 10 (Raspbian Desktop)

mightycandle, 2020-02-22:
Try using  `sudo apt-get install arduino` . This will install arduino IDE. After you write a program and compile it, it might not get transferred to the arduino board. For that , type `sudo chmod --the link given in the error message-- 777`

AtelierMadman, 2020-02-23:
I will give it a try, thank you for the hint