arduino/Arduino #9745: Arduino 1.9.0 Beta shows undefined reference to `loop'/`setup' when upload

panjk0518, 2020-02-11:
Arduino:1.9.0-beta (Windows 10), 开发板："Arduino Leonardo"

D:\Arduino1.9Beta\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino1.9Beta\hardware -hardware C:\Users\潘骏锴\AppData\Local\Arduino15\packages -tools D:\Arduino1.9Beta\tools-builder -tools D:\Arduino1.9Beta\hardware\tools\avr -tools C:\Users\潘骏锴\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino1.9Beta\libraries -libraries C:\Users\潘骏锴\Documents\Arduino\libraries -fqbn=arduino:avr:leonardo -ide-version=10900 -build-path C:\Users\潘骏锴\AppData\Local\Temp\arduino_build_595472 -warnings=none -build-cache C:\Users\潘骏锴\AppData\Local\Temp\arduino_cache_623151 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose -experimental N:\2020潘骏锴\编程\Arduino IDE\hello_world\hello_world.ino
D:\Arduino1.9Beta\arduino-builder -compile -logger=machine -hardware D:\Arduino1.9Beta\hardware -hardware C:\Users\潘骏锴\AppData\Local\Arduino15\packages -tools D:\Arduino1.9Beta\tools-builder -tools D:\Arduino1.9Beta\hardware\tools\avr -tools C:\Users\潘骏锴\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino1.9Beta\libraries -libraries C:\Users\潘骏锴\Documents\Arduino\libraries -fqbn=arduino:avr:leonardo -ide-version=10900 -build-path C:\Users\潘骏锴\AppData\Local\Temp\arduino_build_595472 -warnings=none -build-cache C:\Users\潘骏锴\AppData\Local\Temp\arduino_cache_623151 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose -experimental N:\2020潘骏锴\编程\Arduino IDE\hello_world\hello_world.ino
Using board 'leonardo' from platform in folder: C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2
Using core 'arduino' from platform in folder: C:\Users\潘骏锴\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2
Detecting libraries used...
"C:\\Users\\潘骏锴\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\"" "-IC:\\Users\\潘骏锴\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\Users\\潘骏锴\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\leonardo" "C:\\Users\\潘骏锴\\AppData\\Local\\Temp\\arduino_build_595472\\sketch\\hello_world.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\潘骏锴\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\"" "-IC:\\Users\\潘骏锴\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\Users\\潘骏锴\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\leonardo" "C:\\Users\\潘骏锴\\AppData\\Local\\Temp\\arduino_build_595472\\sketch\\hello_world.ino.cpp" -o "C:\\Users\\潘骏锴\\AppData\\Local\\Temp\\arduino_build_595472\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"D:\Arduino1.9Beta\tools-builder\arduino-preprocessor\0.1.5/arduino-preprocessor" "C:\Users\潘骏锴\AppData\Local\Temp\arduino_build_595472\preproc\ctags_target_for_gcc_minus_e.cpp" "" -- -std=gnu++11
正在编译项目...
"C:\\Users\\潘骏锴\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\"" "-IC:\\Users\\潘骏锴\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\Users\\潘骏锴\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\leonardo" "C:\\Users\\潘骏锴\\AppData\\Local\\Temp\\arduino_build_595472\\sketch\\hello_world.ino.cpp" -o "C:\\Users\\潘骏锴\\AppData\\Local\\Temp\\arduino_build_595472\\sketch\\hello_world.ino.cpp.o"
Compiling libraries...
Compiling core...
Using precompiled core: C:\Users\潘骏锴\AppData\Local\Temp\arduino_cache_623151\core\core_arduino_avr_leonardo_36ec78a83db8af929ce663fd3984b006.a
Linking everything together...
"C:\\Users\\潘骏锴\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega32u4 -o "C:\\Users\\潘骏锴\\AppData\\Local\\Temp\\arduino_build_595472/hello_world.ino.elf" "C:\\Users\\潘骏锴\\AppData\\Local\\Temp\\arduino_build_595472\\sketch\\hello_world.ino.cpp.o" "C:\\Users\\潘骏锴\\AppData\\Local\\Temp\\arduino_build_595472/..\\arduino_cache_623151\\core\\core_arduino_avr_leonardo_36ec78a83db8af929ce663fd3984b006.a" "-LC:\\Users\\潘骏锴\\AppData\\Local\\Temp\\arduino_build_595472" -lm
C:\Users\�˿���\AppData\Local\Temp\cczdGy6A.ltrans0.ltrans.o: In function `main':

C:\Users\�˿���\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\cores\arduino/main.cpp:43: undefined reference to `setup'

C:\Users\�˿���\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\cores\arduino/main.cpp:46: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1
为开发板 Arduino Leonardo 编译时出错。

(Chinese)

per1234, 2020-02-11:
Please post a sketch that causes these errors.

per1234, 2020-02-12:
@panjk0518 we can't see attachments made in email replies. Please go to:
https://github.com/arduino/Arduino/issues/9745
and reply directly to the GitHub issue there.

panjk0518, 2020-02-12:
[sketch_aug20a.zip](https://github.com/arduino/Arduino/files/4189481/sketch_aug20a.zip)

per1234, 2020-02-12:
@panjk0518 which build of Arduino 1.9.0-beta are you using?

@panjk0518's sketch from the sketch_aug20a.zip attached above:
```c++
int val;//定义变量val
int ledpin = 13; //定义数字接口13
void setup()
{
  Serial.begin(9600);//设置波特率为9600，这里要跟软件设置相一致。当接入特定设备（如：蓝牙）时，我们也要跟其他设备的波特率达到一致。
  pinMode(ledpin, OUTPUT); //设置数字13 口为输出接口，Arduino 上我们用到的I/O 口都要进行类似这样的定义。
}
void loop()
{
  val = Serial.read(); //读取PC 机发送给Arduino 的指令或字符，并将该指令或字符赋给val
  if (val == 'R') //判断接收到的指令或字符是否是“R”。
  { //如果接收到的是“R”字符
    digitalWrite(ledpin, HIGH); //点亮数字13 口LED。
    delay(500);
    digitalWrite(ledpin, LOW); //熄灭数字13 口LED
    delay(500);
    Serial.println("Hello World!");//显示“Hello World！”字符串
  }
}
```

panjk0518, 2020-02-12:
I don’t know, yesterday I download it from www.arduino.cc<http://www.arduino.cc>.

panjk0518, 2020-02-12:
I know it now. Build 116.

per1234, 2021-04-02:
Hi @panjk0518. I am not able to reproduce this with the latest hourly build of the Arduino IDE, Arduino IDE beta 1.9 (build 119) or Arduino IDE 2.x.

Are you still having this problem when using the current version of the Arduino IDE?

per1234, 2021-05-15:
Closing due to lack of response.

@panjk0518 if you'll make a reply here to provide the requested information I'll be happy to re-open this issue.

closed 2021-05-15 by per1234