arduino/Arduino #9751: new libraries

rgot-org, 2020-02-12:
Please can you add these libraries 
https://github.com/rgot-org/TheThingsNetwork_esp32
https://github.com/rgot-org/TTN_M5Stack
thanks !

per1234, 2020-02-12:
Hi @rgot-org. In each of the repositories, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

rgot-org, 2020-02-12:
i've made the release 0.0.1

cmaglie, 2020-02-17:
Done.

closed 2020-02-17 by cmaglie