arduino/Arduino #9756: [Library Manager] Please add weatherLocation Library to manager

JHershey69, 2020-02-14:
https://github.com/JHershey69/weatherLocation-Library

cmaglie, 2020-02-17:
Done.

Please note that in your release tagged as `1.0.5` you still have the version `1.0.4` written in library.properties:

https://github.com/JHershey69/weatherLocation-Library/blob/v1.0.5/library.properties

this error will prevent 1.0.5 from being published, to fix this:

1. remove the tag 1.0.5
2. re-do the tag 1.0.5 on the current master

after this, the release 1.0.5 will be automatically published within an hour.

cmaglie, 2020-02-17:
sorry the correct instructions are:

1. remove the tag 1.0.5
2. fix library.properties version to 1.0.5
3. commit & psuh
4. re-do the tag 1.0.5 on the current master

closed 2020-02-17 by cmaglie