arduino/Arduino #9757: [Library Manager] EthernetWebServer library for Ethernet Shields

khoih-prog, 2020-02-14:
https://github.com/khoih-prog/EthernetWebServer

This is simple yet complete WebServer library for `AVR, Teensy, etc.` boards running Ethernet shields. The functions are similar and compatible to ESP8266/ESP32 WebServer libraries to make life much easier to port sketches from ESP8266/ESP32.

The library supports 
1. HTTP Server and Client
2. HTTP GET and POST requests, provides argument parsing, handles one client at a time.

per1234, 2020-02-14:
Hi @khoih-prog. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

khoih-prog, 2020-02-14:
Just done. Thanks

cmaglie, 2020-02-17:
Done.

closed 2020-02-17 by cmaglie