arduino/Arduino #9758: Get error "Could not initialize class com.sun.jna.Native" running arduino_debug.exe

hellyzh, 2020-02-14:
Get error "Could not initialize class com.sun.jna.Native" running arduino_debug.exe for Arduino IDE 1.8.11

````
C:\>"C:\Program Files (x86)\Arduino\arduino_debug.exe"
java.lang.NoClassDefFoundError: Could not initialize class com.sun.jna.Native
        at com.sun.jna.platform.win32.Shell32.<clinit>(Shell32.java:32)
        at com.sun.jna.platform.win32.Shell32Util.getFolderPath(Shell32Util.java:44)
        at com.sun.jna.platform.win32.Shell32Util.getFolderPath(Shell32Util.java:61)
        at cc.arduino.os.windows.Win32KnownFolders.getLocalAppDataFolder(Win32KnownFolders.java:53)
        at processing.app.windows.Platform.recoverSettingsFolderPath(Platform.java:64)
        at processing.app.windows.Platform.init(Platform.java:54)
        at processing.app.Base.<init>(Base.java:207)
        at processing.app.Base.main(Base.java:150)

````

cmaglie, 2020-02-14:
We have released 1.8.12 yesterday may you retry with that one?

matthijskooijman, 2020-02-14:
On IRC, someone reported the same issue with 1.8.10 and 1.8.11 last week, but 1.8.9 would work as expected. They left before I could get some useful info, though.

Looking at my own Arduino install (on Linux), it seems that all com.sun.jna.platform classes come from jna-platform-4.2.2.jar, while the com.sun.jna.Native class comes from jna-4.2.2.jar, so maybe you have the jna-platform-x.x.x.jar file, but not the jna-x.x.x.jar file?

In the zip (windows 1.8.11) I downloaded last week, I see a file "lib/jna-4.2.2.jar". @hellyzh, if 1.8.12 does not fix this for you, maybe you should check if that file is present in your Arduino install? If so, you can see if the file com/sun/jna/Native.class is inside (.jar files are realy just zip files, so if you rename it you can open it as a zip)?

hellyzh, 2020-02-15:
1.8.12 works, thank you.

closed 2020-02-15 by hellyzh