arduino/Arduino #9760: Precompiled libraries don't seem to work in the 1.8.11 version

MiLeG, 2020-02-14:
Hi,
I'm trying to boost my compilation time by including pre-compiled libraries in my project (.a files).

I've managed to acquire the .a files with the "dot_a_linkage=true" option in library.properties, but the "precompiled=true" and "ldflags=-l[library]" don't seem to work.
Even with those lines, the library is still compiled as usual and the *.a files aren't used.

I was following the advice on the Arduino forum posted [here](https://forum.arduino.cc/index.php?topic=653746.msg4407422#msg4407422), as it seems to have worked for some people.

Were there any changes to the build process that are messing with the precompiled library inclusion?

cmaglie, 2020-02-14:
Maybe you put the library in the wrong folder.
Please try with IDE 1.8.12 and enable verbose compile, it should print a message like:

`Library xxxx declared precompiled but folder "yyyy" does not exist`

that should help you to identify the folder `yyyy` where the precompiled file should be placed.

MiLeG, 2020-02-14:
Wow, that was a quick reply and a quick Arduino release!

Yeah, my error is gone and I can use precompiled libraries! Many thanks!
While on the topic of precompiled code, is there any way to include the core.a file as a precompiled core so that it doesn't have to be compiled?

There isn't any guide to doing that, but could I specify the location of the core.a file in the "ldflags" part of the library.properties, so that it will get linked instead of compiling the whole core from scratch?

cmaglie, 2020-02-14:
The `core.a` is already cached, after the first build you should see in the compiler output:

```
Using precompiled core: /tmp/arduino_cache_987594/core/core_arduino_mbed_nano33ble_136dca993952b33a2b70388d9b779ffb.a
```

Also there is no point in adding the `core.a` in the library because the core gets updated independently from the library.

MiLeG, 2020-02-14:
Yeah, including the core in the library doesn't make any sense.

But is it possible to specify a cached core.a file (in the platform.txt file perhaps) to improve the compile time for the 1st compilation.

cmaglie, 2020-02-14:
I don't think it's possible...

(well maybe you can do it if you really try hard hacking the platform.txt, removing all the sources in the core folder so the compiler is tricked to not build anything and produce a new `core.a` without duplicate symbols... I don't know it looks very fragile to me and the build system is not made to really support this use case)

MiLeG, 2020-02-14:
Ok, thanks for the help!

In the end, this seems to have been the old "not running the latest version" type of issue.

closed 2020-02-14 by MiLeG