arduino/Arduino #9761: Please Add my CRMX TimoTwo Wireless DMX library

marccolemont, 2020-02-14:
The link for the library:
https://github.com/marccolemont/CRMX_TimoTwo

cmaglie, 2020-02-17:
Done.

closed 2020-02-17 by cmaglie

marccolemont, 2020-02-17:
Thanks Cristian!

When I make new releases, does that go automatically through the Arduino library system for the users?
Or is each release update a request to be done on the Arduino github like the initial request?

Kind regards / Met vriendelijke groeten,
Marc Colemont   			
MC-Productions <http://www.mc-productions.be/>  	
	
m.	+32 475 73 37 93 
e.	marc.colemont@mc-productions.be <mailto:marc.colemont@mc-productions.be>
w.	www.MC-Productions.be <http://www.mc-productions.be/>
s.	Koosterstraat 22 
B-3570 Alken Belgium 
		 <https://www.linkedin.com/in/marc-colemont-1139bb4/>		 <https://www.facebook.com/mcproductions.be/>	
MC-Productions.be is a brand of Tools4Video BV

If you have received this email in error please notify us back. This message contains confidential information and is intended only for the individual named. If you are not the named addressee you should not disseminate, distribute or copy this e-mail. Please notify the sender immediately by e-mail if you have received this e-mail by mistake and delete this e-mail from your system. If you are not the intended recipient you are notified that disclosing, copying, distributing or taking any action in reliance on the contents of this information is strictly prohibited.
Legal Status: Besloten Vennootschap
Bank: IBAN BE58 0018 7691 9179
Tax-id.No.: 2.296.682.460
VAT Reg.No.: BE 0738.960.450

per1234, 2020-02-17:
@marccolemont the Library Manager indexer job regularly checks all the repositories in the index for new releases. Any release that is compliant with the requirements for addition to the index is automatically added. So there is no need to submit a request for each release.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ