arduino/Arduino #9763: Spanish translation for getting started

Discovertronics, 2020-02-14:
Hi everyone, good afternoon

I am teaching courses on Arduino and it seemed to me that the most appropriate thing was to have the information from the source, so I entered the page and saw that there was very little information in Spanish. That is why I took on the task of translating the content and, based on the Creative Commons Attribution-ShareAlike 3.0 License policies and making a small contribution, I share the translation so that you can redistribute it, this is to your liking.

It is in word 2019 format in case you want to use it for anything

I apologize in advance for the mistakes of any and if someone already did it and I am duplicating, I am new to the forum and the little I surfed I saw nothing similar

Greetings to all!
[Arduino en español.zip](https://github.com/arduino/Arduino/files/4206775/Arduino.en.espanol.zip)

priyanshu1909, 2020-07-02:
i want to work on this issue . Please assign me this task and guide me as i am new to this

Discovertronics, 2020-07-02:
Hi priyanshu1909, thank for your interes

Right know, translation is complete. You are welcome to improve changes in order tor obtain a better translations or better interpretations abou it. Only thing you need to do is publish and notify Original author (arduino.cc)

Thanks again for your interest, I hope this job can be helpfull to you

igmtz, 2023-04-04:
Hey! I would like to contribute in improving translations, is someone currently working on this?

per1234, 2023-04-05:
Hi @igmtz. It won't be possible for Arduino to accept translations of this content because we don't have the necessary localization infrastructure in place to deliver it.

However, I invite you to contribute to the translation of the Arduino IDE user interface text. We have an excellent internationalization and localization system in place in that project that will make it easy for you to translate and ensure we will deliver your contributions to the many thousands of Arduino IDE users quickly and efficiently.

There is information about how to get involved in that initiative here:

https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/translation.md#translator-guide

Please let me know if you have any questions.

igmtz, 2023-04-05:
Thank you @per1234! I already joined the project on transifex, I'm figuring out how the platform works. Do I have to download the doc to translate it or is everything handled on transifex?

per1234, 2023-04-06:
>I already joined the project on transifex,

Nice!

>Do I have to download the doc to translate it or is everything handled on transifex?

Everything is handled on Transifex. We have a system that automatically uploads the Arduino IDE user interface source English text to Transifex. This means the strings available for translation on Transifex are always in sync with the development work on the Arduino IDE codebase.

igmtz, 2023-04-07:
Thank you @per1234 🙌🏼 I already have been translating some strings. How is the process for my changes to be reviewed or validated?